springfall2008/batpred #2466: apps.yaml validation on reserve sensor / Warn: get_state_wrapper - Units mismatch for number.min_soc_on_grid: expected wh, got % after conversion

WyndStryke, 2025-06-16:
**Describe the bug**

apps.yaml validation on reserve sensor

From documentation:

> **Battery Information**
>
> ...
> reserve - sensor name for the reserve setting in %
> ...

```
    # Reserve SoC will be configured with number.min_soc_on_grid
    has_reserve_soc: True
```

    
```
  # Reserve SoC is used for Freeze Charge and Hold Charge modes.  This must never be set to less than 10%
  # (to match 90% DoD) and preferably a little higher to be safe.  
  # Also see the predbat setting input_number.predbat_set_reserve_min which should be set to 10 minimum, 
  # and input_number.predbat_best_soc_min which needs to be set >= kWh equivalent of that.
  reserve:
    - number.min_soc_on_grid

```

The following appears in the log.

```
5321	2025-06-16 15:45:11.858614: Inverter 0 Charge settings: timed charged is disabled, power 0.4 kW
5320	2025-06-16 15:45:11.858566: Inverter 0 charge windows currently []
5319	2025-06-16 15:45:11.858525: Inverter 0 scheduled charge enable is False
5318	2025-06-16 15:45:11.858379: Inverter 0 write_and_poll_switch: No write needed for scheduled_charge_enable as False == False
5317	2025-06-16 15:45:11.858028: Inverter 0 SOC: 14.76kW 89% Current charge rate 400W Current discharge rate 1156W Current battery power 0.0W Current battery voltage 240.4V Grid power 0.0W load power 455.0W PV Power 4784.0W
5316	2025-06-16 15:45:11.857154: Inverter 0 with soc_max 16.58 kWh nominal_capacity 16.58 kWh battery rate raw 11500.0 w charge rate 7.0 kW discharge rate 7.0 kW battery_rate_min 0.0 w ac limit 7.0 kW export limit 5999940.0 kW reserve 15.0 % current_reserve 10.0 % temperature 25 c
5315	2025-06-16 15:45:11.857034: Reserve min: 15.0% Battery_min:10.0%
5314	2025-06-16 15:45:11.856910: Warn: get_state_wrapper - Units mismatch for number.min_soc_on_grid: expected wh, got % after conversion
5313	2025-06-16 15:45:11.856628: Note: Inverter 0: Using inverter type FoxESS - not all features are available on all Inverter types
5312	2025-06-16 15:45:11.856472: Inverter 0: Type FoxESS FoxESS
5311	2025-06-16 15:45:11.715464: Today's actual load so far 2.86 kWh with 0.0 kWh Car/iBoost excluded and 3.98 kWh import ignored.
5310	2025-06-16 15:45:11.715426: Today's predicted so far 2.54 kWh with 0.0 kWh car/iBoost excluded and 3.11 kWh import ignored and 0 forecast extra.
5309	2025-06-16 15:45:11.715297: Today's load divergence 105.79 % in-day adjustment 105.5 % damping 0.95x
5308	2025-06-16 15:45:11.666176: Historical days now [7, 14] weight [1, 1]
5307	2025-06-16 15:45:11.665596: Historical data totals for days [7, 14] are [10.69, 8.42] - min 8.42

```

**Expected behaviour**
I was expecting this to be a % sensor, rather than a Wh sensor.  I suspect that this might just be a validation error rather than it actually requiring Wh, since it does seem to correctly pick up the setting.

**Predbat version**

8.21.1

**Environment details**

- FoxESS KH7 with EC4300-H
- Standard HAOS install

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

N/A

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

[apps (2).yaml.txt](https://github.com/user-attachments/files/20759430/apps.2.yaml.txt)
[predbat (3).log](https://github.com/user-attachments/files/20759431/predbat.3.log)

springfall2008, 2025-06-16:
https://github.com/springfall2008/batpred/releases/tag/v8.21.2

WyndStryke, 2025-06-16:
Looks good to me, many thanks :-)

Very happy about the kW / W auto-conversion too.  It makes the Fox template among many others a lot cleaner to set up.

closed 2025-06-16 by WyndStryke