springfall2008/batpred #2470: Car kWh planned but not appearing in plan (IOG)

pacemaker82, 2025-06-17:
**Describe the bug**

IOG had a planned dispatch at 7.30am today, the plan recognised it as a cheap slot and planned to charge the battery in the slot, but didn't reflect any kWh in the car column.

When the slot arrived, the car kWh column correctly showed something - but that could be related to same bug I have here https://github.com/springfall2008/batpred/issues/2381 as the kWh value is not correct or matching the dispatch.

[image]

Plan before the slot happened:

[image]

Plan during the slot

[image]

Note: I am using the octopus integration, not built in to predbat 

**Expected behaviour**

Car kWh column reflect the accurate values as per the planned dispatches.

**Predbat version**

8.21.2

**Environment details**

- HA GREEN
- Giv Battery 9.5 kWh

**LOG FILE**

[predbat.log.txt](https://github.com/user-attachments/files/20769640/predbat.log.txt)

springfall2008, 2025-06-17:
This maybe fixed by https://github.com/springfall2008/batpred/pull/2471

pacemaker82, 2025-06-17:
thanks - will keep an eye on it

pacemaker82, 2025-08-05:
This has started again since 8.23.1

[image]

chalfontchubby, 2026-09-26:
Hello,
If you are still seeing this we do need to see a log and debug.yaml for the plan, alongside the plan screenshot in order to be able to do anything. If that's not possible we'd best close this until we see the symptoms again.

gcoan, 2026-09-27:
I believe @pacemaker82 has stopped using Predbat.

Set to auto close unless confirmed as still an issue

closed 2026-10-05 by chalfontchubby