springfall2008/batpred #2474: Octopus Agile prices for 23:00-23:59 show as zero.

roddypratt, 2025-06-18:
**Describe the bug**
In the Plan, and Rates chart, the import cost shown is zero. For times after this, the cost is (correctly, I think) inferred from previous day's rates.

[image]

[image]

**Expected behaviour**
Import cost for the next 23:00 and 23:30 slot should be inferred from previous day's cost.

**Predbat version**

8.21.3

**Environment details**

Standard HAOS 2025.6.1 on RPi.
Octopus Energy integration 15.2.0

I'm not using Nordpool future rates.

**Predbat debug yaml file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/20795921/predbat_debug.yaml.txt)

springfall2008, 2025-06-21:
It seems like your rates are coming from the octopus integration, I suspect its reporting the wrong rate for 11pm yesterday but I don't have that in the debug yaml, can you check the data it provides?

roddypratt, 2025-06-21:
> It seems like your rates are coming from the octopus integration, 
Yes.

> I suspect its reporting the wrong rate for 11pm yesterday but I don't have that in the debug yaml, can you check the data it provides?
Let me know exactly what you'd need to see, and when.
Maybe it's a GMT/BST thing? Octopus publish the rates at around 4:30pm each day, data covers a 24 hour period from midnight **GMT**

Before 4:30 (i.e. before octopus release new data) I see the behaviour above. After 430PM, the predbat graphs show the expected data.

springfall2008, 2025-06-22:
Can you find when this next occurs and then capture the rate data from the Octopus event showing if the previous data rate was indeed non-zero at midnight?

[image]

springfall2008, 2025-06-22:
So 

event.octopus_energy_electricity_22e5185605_1417150702004_previous_day_rates

e.g.

[image]

closed 2026-08-28 by chalfontchubby