springfall2008/batpred #2484: Additional IOG slots, but plan also trying to export at the same time

zarch1972, 2025-06-21:
I've just plugged the car in and picked up some early Intelligent Go slots.

[image]

But you can see the plan is also going to try and export during the slots.  Which can't happen as you can't import and export at the same time.  You'd hope it would hold charge (as the batteries are already full)

I noticed this the other evening too when I got some slots at 10:30pm.  It tried exporting then too to no avail.

Any thoughts on anything I may have configured wrong?

Thanks

zarch1972, 2025-06-21:
[predbat.log](https://github.com/user-attachments/files/20847064/predbat.log)

[apps (1).txt](https://github.com/user-attachments/files/20847072/apps.1.txt)
[predbat_plan.txt](https://github.com/user-attachments/files/20847073/predbat_plan.txt)

zarch1972, 2025-06-21:
[predbat_debug (2).txt](https://github.com/user-attachments/files/20847077/predbat_debug.2.txt)

springfall2008, 2025-06-22:
Your car charging planned is set to Zappi or Wallbox and is returning False, implying the car is not plugged in.

But you have 'switch.predbat_octopus_intelligent_ignore_unplugged' set to True, meaning it will ignore the Octopus charging plan if the car is plugged.

You probably need to fix your car charging planned sensor or turn off this toggle.

```
  car_charging_planned:
     - 're:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status)'
     - 're:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status)'
```

closed 2026-08-28 by chalfontchubby