springfall2008/batpred #2494: Octopus free electricity session only for car charging now, not house

jambees, 2025-06-22:
**Describe the bug**
There is an Octopus free electricity session today between 2pm and 3pm and my plan shows that it intends to charge the home battery during this window. However, unfortunately it seems Octopus have changed things a bit this year for free saving sessions. If you are on Octopus Intelligent Go you ONLY get free car charging, home electricity use is charged as normal and not free.

**Expected behaviour**
If the electricity tariff is Octopus intelligent go, during a free electricity session, Predbat should not charge the home battery

**Predbat version**
8.21.7

**Environment details**

- Fox Inverter and home battery
- Standard HAOS installer

**Screenshots**
Chart
[image]

Plan
[image]

**Log file**
Can if required but don't think this is needed.

**Predbat debug yaml file**
Again, can if required but don't think this is needed

Nilogax, 2025-06-22:
See also

https://github.com/springfall2008/batpred/discussions/2453

for issues around determining accurate costs as they're giving a flat rate credit regardless of whether the car charges or not.

robstonham-personal, 2025-07-04:
Don't know if it will help, but for the next session I've just commented out the following line from apps yaml
 
  #octopus_free_session: 're:(event.octopus_energy_([0-9a-z_]+|)_octoplus_free_electricity_session_events)'

Seems to have had the effect I needed, ie not planning to charge battery in the free electricity session hour.

A quick win might be a "Octopus Intelligent Tarrif' switch which then ignores the free electricity session events

Nilogax, 2025-07-05:
With solar + car I've

- commented out the free electricity sensor as above
- set manual force freeze export for the free session so the car doesn't charge from my house battery
- set import rate override to 0 for the free session so car charging doesn't add to daily cost metrics

macray76, 2025-07-05:
I don't have the octopus_free_session line in my apps.yaml but my battery was charging (until I stopped it manually) in today's free electricity session.

Not sure if I haven't specified this variable in my apps.yaml file how predbat can still be using it?

gcoan, 2025-07-05:
Predbat could make decisions about whether to use the free electricity sensor depending on whether the IOG sensors/switches are configured or not, but this starts to take predbat into having to understand how the different tariffs work and that's not ideal.

I think the safest thing is for IOG users to comment out the free electricity sensor from apps.yaml as Octopus have changed the rules this year.

Can add this to the guidance documentation if people concur

Nilogax, 2025-07-17:
Here's the relevant part from the email - I'd say it's pretty clear:

> To summarise however;
> 
> -Your properties usage outside of the vehicle charging is billed at the normal rate for this time, it doesn't count as an off-peak window.
> 
> -You need to plug in your vehicle and we'll give you free charging (even if your vehicle doesn't charge you'll still be credited).
> 
> -Customers should not activate a boost charge during this window.

stevedundee2, 2025-07-17:
@Nilogax I see what you mean, but I don't think that's what actually happens. In the session on 5th July, they refunded me 50p, which they equated to 7.08kWh. That means 7p a unit, which means my supply was being charged at 7p. The detailed bill for that day shows the rate dropping whenever consumption goes up to car-like levels, but I can't remember exactly what time it took place.

I think that paragraph is just badly worded. Your interpretation is the most obvious, but I think they just mean the free hour isn't running at cheap rate as of itself. Or they are completely confused themselves and have said one thing and done another.

As a side note, it was a sunny day so nearly half of my charge came from solar diversion and only the balance from the grid. I didn't take 7.08kWh from them even though that's what they've refunded me.

Steve

stevedundee2, 2025-07-20:
@bfayers 

As you're on IOG and you did not plug in a suitable car, then you'd not get any free session credit and that is consistent with the free sessions table. 

The mystery is the £9.21 credit. My 50p credit is dated 14th July and is only visible in the bill (not as a separate line on the app's transaction view), where it is described as "Free Electricity Reward". Your rebate sounds like it is for something else - does it explain it in your bill?

Steve

bfayers, 2025-07-20:
> The mystery is the £9.21 credit. My 50p credit is dated 14th July and is only visible in the bill (not as a separate line on the app's transaction view), where it is described as "Free Electricity Reward". Your rebate sounds like it is for something else - does it explain it in your bill?

All it says is "Customer reimbursement" whatever that means. Previous Free Electricity ones were "Free Electricity Reward" from last year.

stevedundee2, 2025-07-20:
> Predbat could make decisions about whether to use the free electricity sensor depending on whether the IOG sensors/switches are configured or not, but this starts to take predbat into having to [...]

That sounds reasonable. People on IOG who already use the setting because it was sensible before might be getting caught out though. Is there anyway of flagging up the risk in general terms? Perhaps as part of the next update message? I don't know what's possible.

Nilogax, 2025-08-01:
Email from Octopus - looks like we can re-enable those free electricity sensors and don't need any extra mucking around. 

```
Since restarting our free electricity sessions for the summer, we know some Intelligent tariff customers aren’t 100% happy with the scheme – particularly those with multiple different smart products (like an EV and a battery).
 
We heard you, and we’re changing things up for customers on Intelligent Octopus Go and Intelligent Octopus Flux.

You’ll still get your hour of free Intelligent charging, same as before. But from the next session on, we’ll make any extra household power free as well.

Here's how free sessions will work for you from now on:
Plug your car in and stick to your Intelligent schedule to get an hour’s free charging credit. Don’t worry if your car doesn’t charge during the session – more on this below – as long as you’re plugged in, you’ll get the credit.
For the rest of your home, fill your boots! All the electricity you use above your normal household consumption will be free.
```

bfayers, 2025-08-01:
> Here's how free sessions will work for you from now on:
> Plug your car in and stick to your Intelligent schedule to get an hour’s free charging credit. Don’t worry if your car doesn’t charge during [...]

Sounds like we do have to make sure we plug in our car though? Unless I’m misunderstanding

closed 2026-10-28 by springfall2008