springfall2008/batpred #2499: Search Textbox on the Config Page

bowen159, 2025-06-22:
**Is your feature request related to a problem? Please describe.**
Would it be possible to have a search/find box within the config page to make it easier to find specific options.

**Describe the solution you'd like**
A search text box within the Config page. So as you type the configuration name, it narrows down to just that option. Perhaps also on the log page?

**Describe alternatives you've considered**
Currently, I have to search within chrome which is not ideal.

**Additional context**

[image]

springfall2008, 2025-06-22:
Good idea, will be in the next release

springfall2008, 2025-06-22:
https://github.com/springfall2008/batpred/releases/tag/v8.21.9