springfall2008/batpred #2504: Support for bottlecapdave's Carbon Intensity Integration

markbw999, 2025-06-23:
Hi Trefor and team, the carbon intensity UK integration has been broken for me (and I think others) for a few weeks now. I don't think it's the integration itself, rather the source website which is failing to resolve postcodes when determining the user's region.
Bottlecapdave has created an alternative integration which works for me as when configuring it you get to select your region from a dropdown rather than entering your postcode.
It creates a sensor for the current intensity level and two events for the current and next day forecasts. The data looks good but when I try the event in apps.yaml predbat doesn't handle it.
I asked Dave whether the event could be replaced with a sensor that would work with predbat and this was his response...
"The events cannot be a value of a sensor as the data is too large. My other integration uses this setup and is compatible with predbat, but I'm not sure if there is something special predbat is doing." I assume the other integration he refers to is Octopus.
Could you add support for Dave's integration as an alternative to carbon intensity UK?
Thanks, Mark

springfall2008, 2025-06-29:
I've commented on: https://github.com/BottlecapDave/HomeAssistant-CarbonIntensity/issues/25

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing, as the need is met: Predbat has a built-in Carbon Intensity component (`carbon_postcode` / `carbon_automatic`) that calls the API directly, so no HA integration is needed. See [apps.yaml](https://springfall2008.github.io/batpred/apps-yaml/).

closed 2026-10-19 by chalfontchubby