springfall2008/batpred #2516: Forecast.solar monthly correction factors

andrewmusty, 2025-06-29:
Hi,

I've been trying the new added support for forecast.solar using a professional subscription with 3 sites that seem to work very well. 

I have a true south 3.65kw array on the main house that does not have shading issues and a split 3.65kw on the shed (50% true west and 50% true east. It's ok this time of year when the sun is high but during winter at times the shed get a lot of shading that forecast.solar cannot account for. Would it be possible to add a factor support into Predbat like

January > December
set factors_east = [0, 0.55, 0.65, 0.80, 0.90, 1.0, 1.0, 1.0, 0.95, 0.85, 0.70, 0.60, 0.50]
set factors_west = [0, 0.65, 0.75, 0.85, 0.95, 1.0, 1.0, 1.0, 0.95, 0.90, 0.80, 0.70, 0.60]

springfall2008, 2025-06-29:
I could certainly add support for this in the solcast setup.

However, the Predbat solar calibration should do this automatically on a more dynamic basis, would that not be better?

springfall2008, 2026-10-28:
Probably better to use OpenMeteo

closed 2026-10-28 by springfall2008