springfall2008/batpred #2524: Very slow cycles since early June updates of HAOS and PredBat

spedwards-01, 2025-06-30:
**Describe the bug**
PredBat used to take 4mins-something per cycle.  So a 5 minute time was okay in apps.yaml.

After the recent updates PredBat takes 13-17 minutes per cycle most of the time.  

Today there was a DFS session starting at 6pm.

When I noticed the issue at 18:25 the PB Plan was timed at 17:55 and my batteries were not exporting. Before taking debug logs I went iun the GE app to force discharge the batteries.

By the time I started a computer and returned to Predbat to collect logs there was a new PB Plan dated 18:35.  This timing meant that the whole cycle has taken 40mins.

I think I have the logs of the full 40min cycle.  

I have gone back as far as I could using the current HAOS. Even with v8.20.3 the long delay behaviour is present. But with the 40min cycle today its even worse than normal.

[predbat_debug.log](https://github.com/user-attachments/files/20985393/predbat_debug.log)
[predbat_manual.log](https://github.com/user-attachments/files/20985394/predbat_manual.log)

[predbat_debug.zip](https://github.com/user-attachments/files/20985400/predbat_debug.zip)

Sorry I took both updates in the same day or so. So cannot tell if its PB or PB interacting with HAOS that is at fault.

**Expected behaviour**
A clear and concise description of what you expected to happen.

PredBat should complete its cycle wiithin its configured 5 minutes - just like it did back in May.

**Predbat version**

Working version: At least prior to v8.20.3

Broken version: Everything from 8.20.3 onwards. Including 8.21.11

**Environment details**

Hardkernel M1 

- Inverter and battery setup  Give Energy Gen3 and 2x9.5KWh batteries
- Standard HAOS installer or Docker:  Standard HAOS
- Anything else?  Octopus Energy IOG-EV coupled with 24hr 15p export

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

You can download the logfile from the WebUI on the Dash tab select **predbat_debug.log**

**Predbat debug yaml file**
This is important for any plan related issues.

Once you have captured the issue go to the Web UI, in the Dash tab click on **predbat_debug.yaml** and wait 30 seconds for it to download. Select 'keep' if your Web Browser thinks the file is dangerous.

Rename the download file to predbat_debug.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

spedwards-01, 2025-06-30:
Another, maybe useful tidbit.  Around the time it got slow I configured several alternate tarriffs to check.  I have since removed them from the config by commenting them out.  This made no difference to the speed of the calculation cycle.

gcoan, 2025-07-01:
See also #2394 raised previously about slower performance of predbat, but not as long as 40 minute update cycles.

Having carbon enable turned on does contribute to slower performance, but the other issue has identified that some of the rate optimisation takes a long time

springfall2008, 2025-07-03:
YOu might want to turn off

switch.predbat_calculate_second_pass

Although I doubt it will make a huge difference.

What physical machine are you running on, this took around 6 seconds on my MAC Mini?

spedwards-01, 2025-07-03:
Thank You @gcoan, I had not seen that one. Very interesting.

@springfall2008, Many many thanks for you work on PredBat!

I am running HA on a Hardkernel Odroid M1:  https://www.odroid.co.uk/ODROID-M1-8GB

I now have PB back in its usual timebox, actually a little quicker. Its now doing 3-4 minute cycles. But I have only managed it by turning off pretty much everything I had turned on before the slowdown became visible.

One small change to me had a large effect on timing. Until recently the Carbon sensor in my installation would not survive re-boots. I had to uninstall and reinstall it to resume functionality.  I forgot about it, and that PB was configured to use it.  At some recent point that integration changed to always be working. I had not noticed this until I saw @gcoan's comment in this issue.

With carbon intemsity turned off, the timing dropped to 15-21 mins.  So it was costing around half the elapsed time, perhaps slightly more than half.

I have since been playing with other settings, but do not yet have a chart of the timing differences each makes.

I am continuing to work with the system and will report back as to the amount each system option uses on my installation in a couple of days or so. 

I wonder, does anyone know if there is a application profiler that I could run on Predbat inside the live system?

pacemaker82, 2025-07-04:
@leacho73 try just searching "calculate second pass" without predbat - in case all your entities for predbat start with a different name?

[image]

leacho73, 2025-07-04:
Still no I am afraid @pacemaker82

leacho73, 2025-07-04:
> You need expert mode turned on to see some switches.

Good shout @spedwards-01 - thanks for the reminder, I've just enabled that and now the sensor is show. @springfall2008 - the second pass is already off.

springfall2008, 2025-07-10:
I've done a whole load of optimisations, please feedback:

https://github.com/springfall2008/batpred/releases/tag/v8.22.4

leacho73, 2025-07-11:
8.22.4 installed here and I am still seeing 7-8min runtimes - although better than before still feels like its taking a while to crunch the numbers.

springfall2008, 2025-07-12:
> 8.22.4 installed here and I am still seeing 7-8min runtimes - although better than before still feels like its taking a while to crunch the numbers.

If you share a logfile I can see where the time goes

pacemaker82, 2025-07-12:
Here is a pretty long one for me (I define long that is anything over 5 mins because the every 5 minute run gets delayed, and ends up starting immediately after):

[predbat_long_log.txt](https://github.com/user-attachments/files/21197097/predbat_long_log.txt)

pacemaker82, 2025-07-12:
I was using ChatGPT to analyse my HA Green load, CPU and asked it to look at the code too. It suggested that the amount of historic data that the code is asking for could cause significant delay. So I adjusted my days_previous to only 5 days (reduced from 7) to see what impact, and it looks significant.

I am now seeing 3m40s runs vs 5m, just by reducing the load history by 2 days. Perhaps the HA Green is slow at history data retrieval? Is there a way to cache it as its historic so wont be changing run by run ... ?

```
  days_previous:
    - 2
    - 3
    - 4
    - 5
    - 6
```

pacemaker82, 2025-07-13:
I think the removing load days was a red herring - I've just re-added them in and made no difference to the plan creation time. I will say that the last two versions (22.5 and 22.6) have had significant improvement to my own times, I am now at 3 mins per run which is great compared to over 5 mins in previous versions, and not seen any big spikes yet.

springfall2008, 2025-07-14:
Closing the ticket, please report any new issues

closed 2025-07-14 by springfall2008