springfall2008/batpred #2541: Planning for tariff changing next day

BVGC, 2025-07-06:
I'm trying to figure out a way for predbat to know in advance that I plan to change tariff the next day and to adjust its plan accodingly. Currently it will plan according to the current tariff so behaviour such as exporting in the evening may not be optimal as it assumes the wrong tariff the next day. It also requires me to switch tariff immediately at midnight (or change the yaml/override) otherwise predbat will be assuming the wrong tariff is in operation.

I've looked at using the API override function. This I managed to do fairly successfully for switching to simple tariffs because the rates are stable and predictable. Switching the other way though to Agile i'm struggling to find a way to collect the rates in a format acceptable to predbat. I know the data is available in advance through the octopus api url but is there a way I can splice this and feed to predbat through the api so it only overrides next days pricing?

mchuk, 2025-07-07:
Yes please, this would be awesome!

springfall2008, 2025-08-09:
You can type the rates for the next day into apps.yaml, but its a lot of data for Agile. Better to override to Agile and then type in today's rates for a fixed tariff.

closed 2026-08-28 by chalfontchubby