springfall2008/batpred #2544: Warn: Empty history passed to minute_data, ignoring (check your settings) error in the logs

xhemp, 2025-07-08:
**Describe the bug**
Just wondering where that is coming from, I've checked the solcast entities (pv_forecast_today/tomorrow/d3) and they are all there and got a 31 day history so not sure why I am seeing this.

**Expected behaviour**
Not to have the error in the logs

**Predbat version**

v8.22.2

**Environment details**

Solar Assistant with a Growatt Inverter
HAOS running latest 2025.07.1

**Screenshots**

Screnshot of the forecast sensor showing 7 days:

[image]

**Log file**

[predbat (1).log](https://github.com/user-attachments/files/21119928/predbat.1.log)

**Predbat debug yaml file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/21119933/predbat_debug.yaml.txt)

It's happening on every run and I don't remember changing anything on my side, probably on an upgrade? I don't remember when this started but it's been ongoing for a month or so I'd say.

[image]

Attached are the apps.yaml and the predbat.log[](url)
[predbat (1).log](https://github.com/user-attachments/files/21119835/predbat.1.log)
[apps.yaml.txt](https://github.com/user-attachments/files/21119843/apps.yaml.txt)

gcoan, 2025-07-08:
One thing I noticed in the log file:

2025-07-08 10:28:32.209136: Found 1 inverters totals: min reserve 0.65 current reserve 0.65 soc_max 6.5 soc 2.145 charge rate 0.0 kW discharge rate 3.0 kW battery_rate_min 0.0 w ac limit 3.245 export limit 5999940.0 kW loss charge 3 % loss discharge 3 % inverter loss 4 %

Your export limit looks wrong!

But the error you are getting is from the PV calibration:

2025-07-08 10:30:39.588785: PV Calibration: Fetching PV data for calibration
2025-07-08 10:30:46.210151: Warn: Empty history passed to minute_data, ignoring (check your settings)...
2025-07-08 10:30:46.211914: PV Calibration: Not enough historical data for calibration, only 1 days of history
2025-07-08 10:30:46.259190: PV Calibration: Worst day scaling factor 0.7 best day scaling factor 1.3
2025-07-08 10:30:46.260018: PV Calibration: PV production: 38.17 kWh, Total forecast: 0 kWh adjustment

Looks like predbat can't find your historical pv generation data so I'd look at whatever sensor you have configured in apps.yaml and the history of that sensor. 

Also getting odd message about the Solcast PV data

2025-07-08 10:30:39.527376: PV Forecast today adds up to 55.35 kWh and total sensors add up to 27.68 kWh, factor is 2.0

Worth checking your solcast integration configuration, that you have the half hour forecasts populated ok. Mine doesn't give this warning

xhemp, 2025-07-08:
Thanks for having a look, @gcoan !

I fixed the export limit, it was blank and it doesn't look like it is mandatory but I set it anyway now.

With regards to the historical pv generation data, I am not sure what's the problem there seeing my pv generation data is sensor.growatt_spa_pv_energy and that sensor has a 31 days history, actually I configured sensor.growatt* to have 31 days history data so not sure how it's saying it can't see more than a day, it's weird, unless I am checking in the wrong place?

The PV Forecast is fine on Solcast and even the estimate is quite similar to what I actually get on the day so again, I am not sure how come it's doubling it, that factor 2.0 might be it but I can't see any factor configuration on apps.yaml or config, here's the last 30 days for PV generation:

[image]

I am still re-checking it to see if I am able to see any outlying config.

xhemp, 2025-07-08:
I may have found the issue with the sensor for that line saying 1 day, added sensor.predbat* to recorder so let's see if that will still happen tomorrow.
Still puzzled about the  PV Forecast today issue, though.
I will update here later tomorrow.

xhemp, 2025-07-08:
Yeah, not sure why it's seeing the solcast forecast doubled, today is 25.04 as per Solcast but Predbat is showing 50.08.

[image]

autoSteve, 2025-07-09:
Thanks, @gcoan.

I reckon the BJReplay integration is both "correct" and "wrong", depending on perspective. It's as right as can be in my view.

The intervals retrieved are meansured in average kW expected, not kWh, so summing the `detailedForecast` half-hourly intervals will not yield what might be expected. Each "interval" is kind of like a half-kilowatt-hour expected, so divide each interval by half, and then sum them up to arrive at what might be expected as total generation for a day in kilo-Watt hours. So, each is a per-half-hour average wattage of what might be expected.

At first it's hard to get your head around, but if you always think that expected generation is average Watts and not Watt-hours it makes sense. "Conversion" to Watt-hours will be wrong, but it's a forecast... And it's the closest thing to a daily kWh value that we're going to get.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

gcoan, 2026-08-21:
Hi Rik @chalfontchubby 

The part of the issue:

> Also getting odd message about the Solcast PV data
> 
> ```
> 2025-07-08 10:30:39.527376: PV Forecast today adds up to 55.35 kWh and total sensors add up to 27.68 kWh, factor is 2.0
> ```

is still an active issue I believe.  I see it in my logs:

[image]

and other people have reported it.

Predbat is validating that the total PV forecast and the sum of the half hourly forecasts add up to the same value, and in short, they don't.

The validation sounds sensible but it almost always fails and this confuses people with receiving this warning and we have had other people raising the issue.

I tried to explain above why the figures differ, and as noted, there has been discussion on the Solcast integration on this.

Suggest it does need digging into to work out what to do in Predbat.  I could well be that this validation whilst well meaning will for valid reasons not pass, and so therefore should be removed from Predbat.  That's probably my leaning at the moment but lets dig into it properly and re-read the whole discussion

chalfontchubby, 2026-08-21:
Thanks for confirming @gcoan - keeping this open. Agreed it needs a proper dig into the Solcast total-vs-detailed-forecast mismatch before deciding whether the validation itself is wrong (as you suspect) rather than the data.

chalfontchubby, 2026-08-21:
Dug into this against the discussion above, including autoSteve's explanation on the Solcast side. Confirmed there's no actual scaling bug here - Solcast's `detailedForecast` reports average kW per slot, not kWh, so a naive sum landing at ~2x/4x the sensor's kWh total is expected and already correctly compensated for by Predbat downstream. The real problem was just the log message wording: it labelled the raw kW-sum as "kWh" and phrased factor 2.0/4.0 as if something was wrong, which is what was confusing people.

Reworded the message to say what units were actually detected instead of implying a mismatch - no scaling/logic changes. PR: #4634