springfall2008/batpred #2553: Power Flow - Import from PV to Battery Diagram is Wrong/No House Power

Sephzer, 2025-07-11:
**Describe the bug**
So when charging the battery from solar the power diagram shows the correct arrow and value from PV, however House is sitting at 0W and the arrow from the battery is pointing towards House instead of towards the battery. 

**Expected behaviour**
Home should show house load in W, the arrow for the battery should be reversed when capturing power from solar.

**Predbat version**

8.22.4

**Environment details**

- Inverter: Growatt SPH 5000
- Batteries: Growatt 12kwh (2 x 6kwh)
- Standard HAOS

**Screenshots**

Power Flow
[image]

Solar Assistant Equivalent a minute after Power Flow Screenshot
[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/21181851/predbat.log)

**Predbat debug yaml file**

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/21181857/predbat_debug.yaml.txt)

Sephzer, 2025-07-12:
Looks like I was missing load_power and grid_power.

I've added them and restarted, the house value is now correct however, the grid and battery arrows are the wrong way.

[image]

I am currently exporting as the sun is shining...

springfall2008, 2025-07-12:
Set:

```
   grid_power_invert: True
   battery_power_invert: True
```

Sephzer, 2025-07-13:
Beautiful, this worked a treat. Not sure I've seen this in the documentation anywhere, so apologies if it is, I managed to miss it. Thanks for all the hard work.