springfall2008/batpred #2559: PredBat doesn't start (KeyError: 'stamp') in v8.22.6

murkle, 2025-07-14:
PredBat hasn't been starting up for a few days and I've got this in the logs:

```
2025-07-13 19:23:27.058499: Octopus API: Fetching intelligent dispatches for device 00-1C-55-00-00-05-4C-C7
2025-07-13 19:23:28.525237: Error: Octopus API: 'stamp'
2025-07-13 19:23:28.527407: Error: Traceback (most recent call last):
  File "/config/octopus.py", line 346, in start
    await self.fetch_tariffs(self.tariffs)
  File "/config/octopus.py", line 719, in fetch_tariffs
    await self.clean_url_cache()
  File "/config/octopus.py", line 689, in clean_url_cache
    stamp = self.url_cache[url]["stamp"]
            ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'stamp'
```

I've "fixed" it by commenting out these lines in predbat.py and octopus.py

```
#        if url in self.github_url_cache:
#            stamp = self.github_url_cache[url]["stamp"]
#            pdata = self.github_url_cache[url]["data"]
#            age = now - stamp
#            if age.seconds < (120 * 60):
#                self.log("Using cached GITHub data for {} age {} minutes".format(url, dp1(age.seconds / 60)))
#                return pdata
```

```
#        for url in list(self.url_cache.keys()):
#            stamp = self.url_cache[url]["stamp"]
#            age = now - stamp
#            if age.seconds > (24 * 60 * 60):
#                del self.url_cache[url]
```

gcoan, 2025-07-14:
> OK it won't be related to [#2560](https://github.com/springfall2008/batpred/issues/2560) then

I suggest that it could be the same underlying issue, Octopus has changed the API, the Octopus Integration has been updated (v16) to accept the new API format, but predbat hasn't been (yet)

@murkle swapping to using the Octopus integration may be a workaround in the short term

murkle, 2025-07-22:
@springfall2008 I don't have that file, what should it contain?

I still have the same error with v8.22.7

```
2025-07-21 15:56:23.341316: Octopus API: Found tariff code E-1R-OUTGOING-FIX-12M-19-05-13-E product OUTGOING-FIX-12M-19-05-13 device_id None
2025-07-21 15:56:23.342482: Octopus API: Found tariff code E-1R-INTELLI-VAR-24-10-29-E product INTELLI-VAR-24-10-29 device_id 00-1C-55-00-00-05-4C-C7
2025-07-21 15:56:23.342983: Octopus API: Found tariff code G-1R-OE-LOYAL-FIX-16M-25-02-12-E product OE-LOYAL-FIX-16M-25-02-12 device_id None
2025-07-21 15:56:23.343063: Octopus API: Fetching intelligent dispatches for device 00-1C-55-00-00-05-4C-C7
2025-07-21 15:56:24.939378: Error: Octopus API: 'stamp'
2025-07-21 15:56:24.943307: Error: Traceback (most recent call last):
  File "/config/octopus.py", line 346, in start
    await self.fetch_tariffs(self.tariffs)
  File "/config/octopus.py", line 719, in fetch_tariffs
    await self.clean_url_cache()
  File "/config/octopus.py", line 689, in clean_url_cache
    stamp = self.url_cache[url]["stamp"]
            ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 'stamp'
```

springfall2008, 2025-08-09:
Sorry I mean This looks like a corrupted yaml file in 6adb40d_predbat/cache/octopus.yaml

closed 2026-08-21 by murkle