springfall2008/batpred #2567: Predbat plan defaulting to freeze charge and precenting demand from battery

darryl2k, 2025-07-19:
Over the past couple of days, Predbat status defaults to Freeze Charging. It then also stops the house taking power from the battery in favour of drawing power from the grid. This happens when the batteyr is ata very high charge capacity, so there is no reason not to use it. Also, it experots mpost of the excess solar to the grid rather than charging trhe battery, even when it's low and in some cases is npt able to power the house until 23:30 (IOG low rate)

I expect Predbvat to use battery capacity when avaiable and only to default to grid usage when the battery is depleted. I also expect the battery to charge when there is excess solar, during IOG hours overnoght or when my car is plugged in and I have scheduled low rate hours from IOG.

Predbat version V8.22.6
GivEnergy GIV AC 3.0 inverter and 9.5Kh battery

Standard HAOS installer

[predbat.log](https://github.com/user-attachments/files/21327338/predbat.log)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/21327339/predbat_debug.yaml.txt)

[image]

[image]

[image]

springfall2008, 2025-07-19:
You may want to remove Possible and Moderate from the list?

```
alerts:
    certainty: Possible|Likely|Expected
    event: (Amber|Yellow|Orange|Red).*(Wind|Snow|Fog|Rain|Thunderstorm|Avalanche|Frost|Heat|Coastal
      event|Flood|Forestfire|Ice|Low temperature|Storm|Tornado|Tsunami|Volcano|Wildfire)
    keep: 40
    severity: Moderate|Severe|Extreme
    url: https://feeds.meteoalarm.org/feeds/meteoalarm-legacy-atom-united-kingdom
```

darryl2k, 2025-07-19:
All sorted, thank you.