springfall2008/batpred #2568: Predbat using battery to charge the car

PatrickJanssens, 2025-07-19:
I have the setting "allow car to charge from battery" set to off. This means I expect that when the car is being charged the battery will also be charged and remain at 100% when charged. This has worked well until last night where I noticed throughout the night predbat was discharging the battery to the car, then recharged the battery and back to discharging to the car, on and off. (see screenshots)

[image]

expected behaviour (earlier session on 12/7):

[image]

**Predbat version**
v8.22.6

**Environment details**
I have a GiEnergy 5kW hybrid inverter with 9.8 kWh battery and a 7kW solar panel range.
Standard Home Assistant installation
I'm on Intelligent Octopus Go

HA settings related to car charging:
[image]

Log files:

[predbat.log.txt](https://github.com/user-attachments/files/21327370/predbat.log.txt)

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/21327372/predbat_debug.yaml.txt)

[apps.yaml.txt](https://github.com/user-attachments/files/21327375/apps.yaml.txt)

[predbat_plan.html.txt](https://github.com/user-attachments/files/21327376/predbat_plan.html.txt)

springfall2008, 2025-07-19:
Did you recently upgrade the Octopus Integration, if so something changed and you need the latest predbat (https://github.com/springfall2008/batpred/releases/tag/v8.22.7) to be compatible?

PatrickJanssens, 2025-07-19:
Thanks for your quick  reply! There was an update from the Octopus integration today. My predbat is set to update automatically but I notice it was still on v8.22.6. so manually did the updated now.  So will check if that solves the issue.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

PatrickJanssens, 2026-08-20:
This issue can be closed.

closed 2026-08-20 by gcoan