springfall2008/batpred #2577: Compare not working for : Intelligent GO import

jasonwragg, 2025-07-22:
**Describe the bug**
Intelligent GO import doesnt update at 00:00

**Expected behaviour**
It should populate along with all the other tariffs

**Predbat version**

v8.23.1 (and previous version)

**Environment details**

- Givenergy Inv / Battery / Solar
- HAOS Add-on

**Screenshots**

[image]

**Log file**
[predbat (3).log](https://github.com/user-attachments/files/21373504/predbat.3.log)

You can download the logfile from the WebUI on the Dash tab select **predbat_debug.log**

# Tariff comparison feature
  # Adjust this list to the tariffs you want to compare, include your current tariff also
  # Octopus region code (https://energy-stats.uk/dno-region-codes-explained/)
  octopus_region: "L"
  compare_list:
    - id: "current"
      name: "Current Tariff"
    - id: "cap_seg"
      name: "Price cap import/Seg export"
      rates_import:
        - rate: 27.58
      rates_export:
        - rate: 4.1
    - id: "igo_fixed"
      name: "Intelligent GO import/Fixed export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/INTELLI-BB-VAR-23-03-01/electricity-tariffs/E-1R-INTELLI-BB-VAR-23-03-01-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"
    - id: "igo_agile"
      name: "Intelligent GO import/Agile export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/INTELLI-BB-VAR-23-03-01/electricity-tariffs/E-1R-INTELLI-BB-VAR-23-03-01-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-{octopus_region}/standard-unit-rates/"
    - id: "go_fixed"
      name: "GO import/Fixed export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/GO-VAR-BB-23-02-07/electricity-tariffs/E-1R-GO-VAR-BB-23-02-07-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"
    - id: "go_agile"
      name: "GO import/Agile export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/GO-VAR-BB-23-02-07/electricity-tariffs/E-1R-GO-VAR-BB-23-02-07-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-{octopus_region}/standard-unit-rates/"
    - id: "agile_fixed"
      name: "Agile import/Fixed export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"
    - id: "agile_agile"
      name: "Agile import/Agile export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-{octopus_region}/standard-unit-rates/"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-{octopus_region}/standard-unit-rates/"
    - id: "flux"
      name: "Flux import/Export"
      rates_import_octopus_url: "https://api.octopus.energy/v1/products/FLUX-IMPORT-23-02-14/electricity-tariffs/E-1R-FLUX-IMPORT-23-02-14-{octopus_region}/standard-unit-rates"
      rates_export_octopus_url: "https://api.octopus.energy/v1/products/FLUX-EXPORT-BB-23-02-14/electricity-tariffs/E-1R-FLUX-EXPORT-BB-23-02-14-{octopus_region}/standard-unit-rates"
#    - id: "cosy_fixed"
#      name: "Cosy import/Fixed export"
#      rates_import_octopus_url: "https://api.octopus.energy/v1/products/COSY-22-12-08/electricity-tariffs/E-1R-COSY-22-12-08-{octopus_region}/standard-unit-rates"
#      rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"
#    - id: "cosy_agile"
#      name: "Cosy import/Agile export"
#      rates_import_octopus_url: "https://api.octopus.energy/v1/products/COSY-22-12-08/electricity-tariffs/E-1R-COSY-22-12-08-{octopus_region}/standard-unit-rates"
#      rates_export_octopus_url: "https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-{octopus_region}/standard-unit-rates/"
#    - id: "snug_fixed"
#      name: "Snug import/Fixed export"
#      rates_import_octopus_url: "https://api.octopus.energy/v1/products/SNUG-24-11-07/electricity-tariffs/E-1R-SNUG-24-11-07-{octopus_region}/standard-unit-rates/"
#     rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"

SwiftRR, 2025-07-22:
There is a more recent URL to the IGO API:
```
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/INTELLI-VAR-24-10-29/electricity-tariffs/E-1R-INTELLI-VAR-24-10-29-{octopus_region}/standard-unit-rates'
```

I had the same issue and was using the old URL.
This fixed it for me.

Rob

jasonwragg, 2025-07-22:
Yep that worked!

gcoan, 2025-07-22:
@jasonwragg I have this change to the URL in a forthcoming PR