springfall2008/batpred #2578: Incorrectly shows price override against every import price

gcoan, 2025-07-23:
**Describe the bug**
Octopus power up today, I have set a price override via the manual API, and that is showing correctly as 0p, but all other import prices today are also shown as overridden when they are not.

**Expected behaviour**
Only the prices that are overridden are highlighted with the +- symbol

**Predbat version**
8.20.7

**Environment details**
HAOS, HA 2025.7.3

**Screenshots**
Only a 2 hour rate has been overridden:
[image]

But all import rates show the +/-:
[image]

**Log file**

**Predbat debug yaml file**

springfall2008, 2025-08-16:
I'm wondering if you want to move to the new price override manual API instead?

gcoan, 2025-09-04:
> I'm wondering if you want to move to the new price override manual API instead?

Hi Trefor

First power up event since I raised this, a 2 hour event from 12-2pm tomorrow, so tried to use the new price override manual API.

First problem is that select.predbat_manual_import_rates is only populated 18 hours or so ahead.  The notification from Octopus came this afternoon so when I tried to select the import rate overrides I couldn't because the times weren't populated far enough ahead so I'd have to wait to the evening to be able to do so which is inconvenient (plus needs me to remember it).

Second problem is that this control is fine for making a single half hour slot change but the power up event is for 2 hours and I have the rate override information as a start and end time that I want to push through to predbat via an automation script.
Splitting the time into half hour slots and repeatedly setting the select.predbat_manual_import_rates is cumbersome and non-trivial  scripting!

So the way I do the price overrides using the select.predbat_manual_api and setting rates_import_override is still the preferred approach for me.

Still same formatting problem with the plan on predbat 8.24.2:

[image]

I think its just a plan display problem

gcoan, 2026-08-28:
yep, working now, thanks Rik