springfall2008/batpred #2579: Using Apps tab to update days_previous_weight saves value as string

snarfattack, 2025-07-23:
**Describe the bug**
I used the Apps ui to update days_previous_weight to a different value, Instead of being saved as an integer in the apps.yaml file, it was saved as a string with single quotes around it. On restart, Predbat throws several "non-int of type 'float'" errors

**Expected behaviour**
Value should be saved as in int in apps.yaml, without the single quotes around it.

**Predbat version**
v8.23.1

**Environment details**
- Inverter and battery setup: N/A
- Standard **HAOS installer**
- Anything else?

**Screenshots**
N/A

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

14843 | 2025-07-23 17:47:24.237825: Error: can't multiply sequence by non-int of type 'float'
14842 | 2025-07-23 17:47:24.237471: Info: record_status Error: Exception raised can't multiply sequence by non-int of type 'float'
14840 | TypeError: can't multiply s equence by non-int of type 'float'
14826 | 2025-07-23 17:47:24.200864: Error: Traceback (most recent call last):
14825 | 2025-07-23 17:47:24.191875: Error: Exception raised can't multiply sequence by non-int of type 'float'

**Predbat debug yaml file**
This is important for any plan related issues.

N/A

springfall2008, 2025-08-06:
https://github.com/springfall2008/batpred/pull/2593

springfall2008, 2025-08-07:
Fix has been released

closed 2025-08-07 by springfall2008