springfall2008/batpred #2584: Predbat doesn't call the `discharge_start_service` when export starts

iangregory, 2025-07-28:
**Describe the bug**
When export is due to start, 

For example, at 21:

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

xxxx

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

You can download the logfile from the WebUI on the Dash tab select **predbat_debug.log**

**Predbat debug yaml file**
This is important for any plan related issues.

Once you have captured the issue go to the Web UI, in the Dash tab click on **predbat_debug.yaml** and wait 30 seconds for it to download. Select 'keep' if your Web Browser thinks the file is dangerous.

Rename the download file to predbat_debug.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

iangregory, 2025-07-28:
Scheduling a manual export does correctly call `discharge_start_service` (at 10:33:16)

[predbat(2).log](https://github.com/user-attachments/files/21466227/predbat.2.log)

CossieRob, 2025-08-03:
I think I am having similar issues with my Fox H1-G2 inverter. It only seems to happen, that I've noticed, when PB is cycling the battery due to cheaper rates. Sometime is fails to change from charge to discharge or vice versa. Manual controls still work.

[predbat.log](https://github.com/user-attachments/files/21565082/predbat.log)

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/21565088/predbat_debug.yaml.txt)

springfall2008, 2025-08-10:
> Scheduling a manual export does correctly call `discharge_start_service` (at 10:33:16)
> 
> [predbat(2).log](https://github.com/user-attachments/files/21466227/predbat.2.log)

The logfile clearly says the service was called:

2025-07-28 10:33:16.146846: Inverter 0 Calling service discharge_start_service domain discharge service_name input_select/select_option with data {'entity_id': 'input_select.predbat_requested_mode', 'option': 'Discharging'}

Is predbat_requested_mode some template sensor you created?

iangregory, 2025-08-11:
Here's two examples, a non-working and working one, from yesterday's runs. The log lines beginning `####` is additional logging that I've added. 

**Working**
```bash
$ egrep 'adjust_export_immediate\(|Exporting|discharge_start' *.log
predbat.6.log:2025-08-10 20:32:30.584606: Exporting now - current SoC 13.427 and target 8.74 power adjust 1.0
predbat.6.log:2025-08-10 20:32:42.732841: Inverter 0 write_and_poll_value: Wrote 20:30:00 to discharge_start_time, successfully now 20:30:00
predbat.6.log:2025-08-10 20:33:32.906926: #### adjust_export_immediate(target_soc=65, freeze=False) called from: execute_plan in /config/execute.py:441
predbat.6.log:2025-08-10 20:33:32.947455: #### call_service_template(service=discharge_start_service,data={'device_id': '', 'target_soc': 65, 'power': 6950},domain=discharge) called from: adjust_export_immediate in /config/inverter.py:2236
predbat.6.log:2025-08-10 20:33:32.947602: Inverter 0 Calling service discharge_start_service domain discharge service_name input_select/select_option with data {'entity_id': 'input_select.predbat_requested_mode', 'option': 'Discharging'}
predbat.6.log:2025-08-10 20:33:47.534769: Completed run status Exporting
predbat.6.log:2025-08-10 20:33:47.764798: Info: record_status Exporting target 100%-73%
```

**Non-working**
```bash
$ egrep 'adjust_export_immediate\(|Exporting|discharge_start' *.log
predbat.5.log:2025-08-10 21:46:51.403024: Exporting now - current SoC 12.674 and target 0.54 power adjust 1.0
predbat.5.log:2025-08-10 21:46:59.479883: Inverter 0 write_and_poll_value: Wrote 21:45:00 to discharge_start_time, successfully now 21:45:00
predbat.5.log:2025-08-10 21:47:41.619021: #### adjust_export_immediate(target_soc=0, freeze=False) called from: execute_plan in /config/execute.py:441
predbat.5.log:2025-08-10 21:47:55.999987: Completed run status Exporting
predbat.5.log:2025-08-10 21:47:56.326995: Info: record_status Exporting target 94%-4%
predbat.5.log:2025-08-10 21:51:51.288372: Exporting now - current SoC 12.634 and target 0.54 power adjust 1.0
```

The difference seems to be that `adjust_export_immediate` is being called with a target SOC of 0 in the second case, and a [target soc of zero causes the discharge stop service to be called](https://github.com/springfall2008/batpred/blob/6fc24fd8b532117c81c20e347f37cd673ed25883/apps/predbat/inverter.py#L2213)

So I think what might be happening is that predbat is calculating an `export_limits_best[0]` of 0 - presumably a full discharge, but then 0 is special cased later in the code to stop discharging.

However, I increased `battery_min_soc` to before the tests above 4 which I guessed would mean that target_soc would never be 0 - this didn't seem to help.

@springfall2008 does this sound plausble? Am I on the right track?

springfall2008, 2025-08-16:
Fix on 'main' for testing