springfall2008/batpred #2587: Import rate changes to intelligent rate for current timeslot when non intelligent car is charging.

tractor137, 2025-08-01:
**Describe the bug**
Whenever my non intelligent car is plugged in and drawing enough power to turn on my 'car charging now' sensor the plan seems to assume that it must be an intelligent slot and overrides the rate for that half hour period to the low rate (i.e. 7p/kWh). 

**Expected behaviour**
I'd expect the import rate to be drawn only from the octopus current rate sensor, however this doesn't appear to be the case. Current rate sensor and log show that it knows the rate is high, yet the html plan appears to show a low rate because a car is plugged in and charging despite it not being an intelligent slot. 

**Predbat version**

v8.23.1

[predbat-2.log](https://github.com/user-attachments/files/21544590/predbat-2.log)

[image]

**Environment details**

Standard install. Givenergy 5kw gen 1 with 9.5kwh battery

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

You can download the logfile from the WebUI on the Dash tab select **predbat_debug.log**

**Predbat debug yaml file**
This is important for any plan related issues.

Once you have captured the issue go to the Web UI, in the Dash tab click on **predbat_debug.yaml** and wait 30 seconds for it to download. Select 'keep' if your Web Browser thinks the file is dangerous.

Rename the download file to predbat_debug.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

gcoan, 2025-08-03:
It sounds like you need to automate your second (dumb) car to charge in alignment to what your main car and predbat is doing

You could use the Octopus intelligent despatching sensor to determine when its an Octopus cheap period https://bottlecapdave.github.io/HomeAssistant-OctopusEnergy/entities/intelligent/#is-dispatching

Or alternatively use Predbat's binary sensors that indicate importing and exporting https://springfall2008.github.io/batpred/output-data/#battery-status

chalfontchubby, 2026-08-29:
Thanks for the report. This issue has been open for over 6 months with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-09-05).

chalfontchubby, 2026-09-09:
Closing as flagged - no response to the note above, so we're assuming this is no longer live.

If you're still hitting this, please reopen or open a fresh issue with a current version number and a `predbat_debug_*.yaml` captured while it's happening; we can replay that directly against current code.

_Posted by Claude on Rik's behalf._

closed 2026-09-09 by chalfontchubby