springfall2008/batpred #2598: Car charging cost not adding to current days total cost

jasonwragg, 2025-08-07:
**Describe the bug**
Charging slows on IOG are not adding to the total cost for the day

**Expected behaviour**
Total for the day should include car charging costs

**Predbat version**

8.23.3

**Environment details**

- GivEnergy Gen3 Battery, GE 3.6 Inverter, Solar, EV, Zappi
- Standard HOAS

**Screenshots**

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

You can download the logfile from the WebUI on the Dash tab select **predbat_debug.log**

**Predbat debug yaml file**
Attached

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/21667772/predbat_debug.yaml.txt)

[predbat_plan.html.txt](https://github.com/user-attachments/files/21667775/predbat_plan.html.txt)

Once you have captured the issue go to the Web UI, in the Dash tab click on **predbat_debug.yaml** and wait 30 seconds for it to download. Select 'keep' if your Web Browser thinks the file is dangerous.

Rename the download file to predbat_debug.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

jasonwragg, 2025-08-07:
[image]

springfall2008, 2025-08-09:
So this is because 'number.octopus_energy_a_cac718e1_intelligent_charge_target' is assigned to car_carhging_limit and its set to 11.7 which is the cars current state of charge, hence while octopus gave you slots the car shouldn't charge any futher.

If this limit is broken then I'd suggest just setting this to the max battery size (or comment it out).

springfall2008, 2025-08-16:
I think its best to comment out, I think as Octopus doesn't know the current battery level its treated as a percentage to add.

If you set 'car_charging_soc' to 0 then it should kind of work as expected.

jasonwragg, 2025-08-17:
I could create a template sensor to show the target charge in % as i have that info, would that work?

chalfontchubby, 2026-09-23:
@jtrotman-hf I suspect your comment is a bit tangential (although related) to the earlier issue, which was heading towards resolved. 

I'd suggest you raise a new issue providing your debug.yaml and perhaps a log. 
I have Iog and car outside the inverter ct clamp, although almost never charge the car from solar since it's rare that export is lower than my import rate.

chalfontchubby, 2026-09-23:
@jasonwragg Could you confirm what the status is after the last question a year ago? Is there a remaining request?

closed 2026-10-05 by chalfontchubby