springfall2008/batpred #2608: New install Error is startup log - can't match Octopus Intelligent Sensors

Sycamore967, 2025-08-10:
Describe the bug
I've decided to rebuild a new HA and install new up to date versions of everything. Versions below. Prebat starts but has these errors listed in the log against the addon:

2025-08-10 20:40:02.754268: Warn: Regular expression argument: octopus_intelligent_slot unable to match re:(binary_sensor.octopus_intelligent_slot|re:binary_sensor.octopus_energy_intelligent_dispatching), now will disable
2025-08-10 20:40:02.757132: Warn: Regular expression argument: octopus_ready_time unable to match re:(time.octopus_energy_([0-9a-z_]+|)_intelligent_target_time), now will disable
2025-08-10 20:40:02.760911: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
2025-08-10 20:40:02.764087: Warn: Regular expression argument: octopus_free_session unable to match re:(event.octopus_energy_([0-9a-z_]+|)_octoplus_free_electricity_session_events), now will disable
2025-08-10 20:40:05.348431: Warn: Validation of apps.yaml found configuration item 'car_charging_energy' element sensor.hypervolt_session_energy_total_increasing is not a valid type float
2025-08-10 20:40:05.349236: Warn: Validation of apps.yaml found configuration item 'car_charging_now_response' element 88 is not a string
2025-08-10 20:40:05.349831: Error: Validation of apps.yaml found 2 configuration errors
Watching ['./energydataservice.py', './unit_test.py', './db_manager.py', './ha.py', './alertfeed.py', './web.py', './utils.py', './prediction.py', './plan.py', './gecloud.py', './hass.py', './predheat.py', './solcast.py', './inverter.py', './apps.yaml', './output.py', './plugin_system.py', './octopus.py', './compare.py', './download.py', './predbat.py', './config.py', './db_engine.py', './fetch.py', './execute.py', './userinterface.py', './futurerate.py'] for changes
2025-08-10 20:40:08.992657: Warn: Validation of apps.yaml found configuration item 'car_charging_energy' element sensor.hypervolt_session_energy_total_increasing is not a valid type float
2025-08-10 20:40:08.993006: Warn: Validation of apps.yaml found configuration item 'car_charging_now_response' element 88 is not a string
2025-08-10 20:40:08.994669: Error: Validation of apps.yaml found 2 configuration errors

The carbon intensity one I'm expecting - I can't get it to configure properly so I may switch to bottle cap daves one.
The hypervolt one is annoying but also expected due to the 'unknown' value when the car isn't actually charging.

The reset, as this is a clean install, I don't understand. I tried to resolve the first one by amending apps.yaml to include the actual sensor value but that failed, so I'm logging this bug.

Expected behaviour
Clean startup without errors.

Predbat version: 8.23.5
GivTCP v3.1.6
Octopus Energy (BCD) v16.0.2 installed with defaults

Environment details

GE 3.6 qty 2 9.5 batteries
Standard HAOS installer
Started from clean build of HA, installed GivTCP, MQTT , Hypervolt and then Predbat
Screenshots
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

Log file

[predbat (1).zip](https://github.com/user-attachments/files/21707314/predbat.1.zip)

gcoan, 2025-08-10:
Obvious question, but you have installed the Octopus integration and configured it and enabled all the sensors predbat requires as per the installation instructions?

Sycamore967, 2025-08-16:
It took a while but things seemed to sort themselves out as each sensor actually saw some data

closed 2025-08-16 by Sycamore967