springfall2008/batpred #2624: Add support for Kostal Plenticore inverter to Predbat Docu

mbuhansen, 2025-08-19:
I have now made a description on my setup to the kostal Plenticore inverter.

This should work with both Plenticore G1/G2 and G3 inverters

I have attached 2 files, on with setup i HA, and the apps yaml file i use.
I hope you can use it on Predbat documentation to other kostal users.

[Kostal.setup.HA.txt](https://github.com/user-attachments/files/21862868/Kostal.setup.HA.txt)

[Apps.txt](https://github.com/user-attachments/files/21862869/Apps.txt)

gcoan, 2025-12-09:
Thank you @mbuhansen for sharing this, I have merged it into the predbat documentation, sorry for the time it has taken to do this.

If you could check it for any errors or anything that has changed since you shared the config; I notice there are some service calls commented out for example in the apps.yaml template?

Kostal apps.yaml: https://github.com/gcoan/batpred/blob/main/templates/kostal.yaml

Inverter setup: https://github.com/gcoan/batpred/blob/main/docs/inverter-setup.md#kostal-plenticore

Your notes were very comprehensive, thank you.  I only had one question on the four input_numbers to be created, what is the max and minimum values of these input_numbers ?

mbuhansen, 2025-12-11:
I have now update both paste bin links.

In kostal setup i have added 2 new input_number i have forgot i first version, and min/max value is added. I have also added a new automation i'm runnig with now.

Apps.yaml is update. I have change entinities name so it is the same as name that G1 and G2 inverter use. 

Yes it is right charge freeze service is comment out in apps yaml, it can be use but i dont use it, becuse predbat make alot of change between charge start and charge freeze service, when predbat want to hold charge. I don't know why predbat does this .

gcoan, 2025-12-11:
> Yes it is right charge freeze service is comment out in apps yaml, it can be use but i dont use it, becuse predbat make alot of change between charge start and charge freeze service, when predbat want to hold charge. I don't know why predbat does this .

Thank you, I have merged this into the documentation

I'm not quite sure I understand the problem?

mbuhansen, 2025-12-12:
I dont now what the exactly  problem is, but when predbat want to hold charge, it could switch every 5 min betweenuse  charge start service (hold charge) , and charge freeze service. 
Chatgpt said something about the difference between choosing Hold charge or charge freeze being too small, but didn't see much improvement in using charge freeze instead of hold charge. So i disable it, but it is long time ago now.

mbuhansen, 2025-12-14:
Hi.

please update automation in documentation again, i have remove a function i thought was working with car hold charging function but it does not. and this made some trouble with the charge and hold charge function, it is now working correct again.

gcoan, 2025-12-14:
r/e your charge freeze behaviour, there might be something set wrongly.  If you want to log a new issue in this with logfile and debug file we can have a look see what it is

here's my plan execution from this morning, as you can see it went from Charging to Hold Charge for 5 minutes and then Freeze Charge for several hours

[image]

In practical terms there isn't much difference between Freeze and Hold charge, both are holding the charge at the current level, preventing discharge and running off grid import.  It's used when import rates are low but Predbat doesn't want to charge your battery up any more.

My battery has a battery pause mode which Predbat used, stopping discharge from 4am, so no change in setting between Hold and Freeze charge
[image]

if your inverter doesn't have battery pause then predbat will set the charge rate to zero on both freeze charge and hold charge.

I'll update the documentation, will ping when done

gcoan, 2025-12-16:
> please update automation in documentation again, i have remove a function i thought was working with car hold charging function but it does not. and this made some trouble with the charge and hold charge function, it is now working correct again.

I've done that.  Hopefully all done now.

One thing I did notice is that in the apps.yaml you have some of the sensor names in mixed upper and lower case.  Home Assistant sensors are always lower case and I've seen certainly one issue caused by case being wrong.  I've amended the apps.yaml template to change the few entries that were affected to lower case.

Please do raise a separate issue with logs etc for when you've looked at charge freeze again

closed 2025-12-16 by gcoan