springfall2008/batpred #2642: bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled

kitenski, 2025-09-01:
**Describe the bug**
I see a string of bad REST data messages in the log every hour.

**Expected behaviour**
When loading the page it appears to load and read fine, so wouldn't expect these warnings.

**Predbat version**

1.2.9

**Environment details**

GivEnergy 3kW AC Coupled Inverter Charger
GivEnergy 9.5kWh Gen 3 LiFePO4 Battery
Standard HAOS install

20499 | 2025-09-01 10:25:57.818567: Completed run status Freeze exporting with Errors reported (check log)
-- | --
20328 | 2025-09-01 10:25:16.927946: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
20327 | 2025-09-01 10:25:16.918272: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
20227 | 2025-09-01 10:25:01.472442: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.79 kWh to make new average 20.86 kWh (percent 100%)
20179 | 2025-09-01 10:20:17.210093: Completed run status Freeze exporting with Errors reported (check log)
20170 | 2025-09-01 10:20:17.141710: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
20169 | 2025-09-01 10:20:17.132159: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
20069 | 2025-09-01 10:20:01.905779: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.74 kWh to make new average 20.81 kWh (percent 100%)
20021 | 2025-09-01 10:15:48.700567: Completed run status Freeze exporting with Errors reported (check log)
19850 | 2025-09-01 10:15:17.554241: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
19849 | 2025-09-01 10:15:17.544593: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
19749 | 2025-09-01 10:15:01.864256: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.67 kWh to make new average 20.74 kWh (percent 100%)
19701 | 2025-09-01 10:10:17.175204: Completed run status Freeze exporting with Errors reported (check log)
19692 | 2025-09-01 10:10:17.106745: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
19691 | 2025-09-01 10:10:17.097352: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
19591 | 2025-09-01 10:10:02.007492: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.82 kWh to make new average 20.89 kWh (percent 100%)
19272 | 2025-09-01 10:05:01.975356: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 21.0 kWh to make new average 21.07 kWh (percent 100%)
19224 | 2025-09-01 10:00:16.759448: Completed run status Freeze exporting with Errors reported (check log)
19215 | 2025-09-01 10:00:16.689799: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
19214 | 2025-09-01 10:00:16.680395: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
19114 | 2025-09-01 10:00:01.996137: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 21.0 kWh to make new average 21.07 kWh (percent 100%)
18795 | 2025-09-01 09:55:01.939046: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.95 kWh to make new average 21.02 kWh (percent 100%)
18747 | 2025-09-01 09:50:17.719237: Completed run status Freeze exporting with Errors reported (check log)
18738 | 2025-09-01 09:50:17.650206: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
18737 | 2025-09-01 09:50:17.640975: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
18637 | 2025-09-01 09:50:02.043654: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.89 kWh to make new average 20.97 kWh (percent 100%)
18589 | 2025-09-01 09:45:47.824612: Completed run status Freeze exporting with Errors reported (check log)
18418 | 2025-09-01 09:45:16.877451: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
18417 | 2025-09-01 09:45:16.868145: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
18317 | 2025-09-01 09:45:01.098171: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.89 kWh to make new average 20.97 kWh (percent 100%)
18269 | 2025-09-01 09:40:16.878901: Completed run status Freeze exporting with Errors reported (check log)
18260 | 2025-09-01 09:40:16.811456: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
18259 | 2025-09-01 09:40:16.802128: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
18159 | 2025-09-01 09:40:01.899553: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.88 kWh to make new average 20.95 kWh (percent 100%)
18111 | 2025-09-01 09:35:46.538268: Completed run status Freeze exporting with Errors reported (check log)
17940 | 2025-09-01 09:35:15.890160: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
17939 | 2025-09-01 09:35:15.880792: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
17839 | 2025-09-01 09:35:01.094481: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.85 kWh to make new average 20.92 kWh (percent 100%)
17791 | 2025-09-01 09:30:18.874399: Completed run status Freeze exporting with Errors reported (check log)
17782 | 2025-09-01 09:30:18.806893: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
17781 | 2025-09-01 09:30:18.797149: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
17681 | 2025-09-01 09:30:03.227601: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.89 kWh to make new average 20.97 kWh (percent 100%)
17631 | 2025-09-01 09:25:34.695291: Completed run status Freeze exporting with Errors reported (check log)
17606 | 2025-09-01 09:25:34.592089: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17605 | 2025-09-01 09:25:34.583088: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17599 | 2025-09-01 09:25:34.570568: Warn: Inverter control auto restart trigger: REST read failure command []
17598 | 2025-09-01 09:25:34.570459: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17597 | 2025-09-01 09:25:34.560678: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17372 | 2025-09-01 09:25:02.094937: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17371 | 2025-09-01 09:25:02.082848: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17365 | 2025-09-01 09:25:02.067453: Warn: Inverter control auto restart trigger: REST read failure command []
17364 | 2025-09-01 09:25:02.067083: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
17363 | 2025-09-01 09:25:02.053557: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
[... 32 lines cut ...]
14810 | 2025-09-01 08:30:02.008741: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.85 kWh to make new average 20.92 kWh (percent 100%)
14763 | 2025-09-01 08:25:42.664095: Completed run status Freeze exporting with Errors reported (check log)
14594 | 2025-09-01 08:25:16.174955: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
14593 | 2025-09-01 08:25:16.165556: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
14493 | 2025-09-01 08:25:01.305611: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.84 kWh to make new average 20.92 kWh (percent 100%)
14445 | 2025-09-01 08:20:18.107674: Completed run status Freeze exporting with Errors reported (check log)
14436 | 2025-09-01 08:20:18.038215: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
14435 | 2025-09-01 08:20:18.028779: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
14335 | 2025-09-01 08:20:01.651146: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.85 kWh to make new average 20.92 kWh (percent 100%)
14017 | 2025-09-01 08:15:02.031108: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 20.84 kWh to make new average 20.91 kWh (percent 100%)
[... cut ...]
13552 | 2025-09-01 08:05:01.447009: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
13546 | 2025-09-01 08:05:01.435205: Warn: Inverter control auto restart trigger: REST read failure command []
13545 | 2025-09-01 08:05:01.435128: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
13544 | 2025-09-01 08:05:01.426207: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled

gcoan, 2025-09-01:
Have you looked at the givtcp logs?

It is very likely this is the same givtcp error that others (including me) have reported https://github.com/britkat1980/giv_tcp/issues/389

For me, I can't use REST at all, have had to comment it out of apps.yaml, as I get an error every single REST request.   Some people have reported that turning Real Time Control off and then On again in the GivEnergy portal has cured the problem, but it didn't for me.

I think the inverter is returning data that givtcp doesn't like, but beyond that, not been able to identify the givtcp issue

kitenski, 2025-09-01:
> What do you see in the givtcp log, is it similar to the other error I linked above?

Thanks for following it, the log looks different:

`2025-09-01 15:50:03,916 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 15:50:06,418 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:07,897 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:07,908 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 15:50:10,392 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:10,625 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:10,636 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 15:50:13,118 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:13,358 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:13,369 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 15:50:15,835 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:17,115 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:50:17,127 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 15:55:03,253 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:55:03,459 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:55:03,471 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 15:55:06,223 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:55:06,467 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 15:55:06,478 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:00:13,301 - GivTCP - write       -  [INFO    ] - Setting Charge Slot 1 to: 16:00 - 16:30 was a success
2025-09-01 16:00:13,766 - GivTCP - write       -  [INFO    ] - Setting Charge Schedule to enable was a success
2025-09-01 16:00:14,742 - GivTCP - write       -  [INFO    ] - Setting Eco mode was a success
2025-09-01 16:00:15,240 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:15,450 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:15,461 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:00:17,933 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:18,206 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:18,217 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:00:20,693 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:20,943 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:20,954 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:00:23,431 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:23,691 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:23,702 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:00:26,208 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:26,473 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:00:26,484 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:05:03,511 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:03,761 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:03,772 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:05:06,257 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:06,500 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:06,512 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:05:08,981 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:09,240 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:09,252 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
[... cut ...]
2025-09-01 16:05:16,714 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:16,714 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2025-09-01 16:05:16,726 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 699)
2025-09-01 16:05:56,544 - GivTCP - write       -  [INFO    ] - Setting Charge Schedule to disable was a success
2025-09-01 16:05:57,012 - GivTCP - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success`

gcoan, 2025-09-01:
See #1637 where the same error messages are reported in givtcp when predbat tries to set battery pause times and can't on an AC3

FileGo, 2025-09-27:
I'm seeing similar issues with GivTCP 3.2 and 9.5 kWh battery and 3 kW AC coupled inverter:

Predbat logs:
```
24541 | 2025-09-27 14:45:31.784180:  Completed run status Charging, Hold for car with Errors reported (check log)
24510 | 2025-09-27 14:45:21.605243:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24509 | 2025-09-27 14:45:21.586834:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24503 | 2025-09-27 14:45:21.575117:  Warn: Inverter control auto restart trigger: REST read failure command []
24502 | 2025-09-27 14:45:21.575017:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24501 | 2025-09-27 14:45:21.562733:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24277 | 2025-09-27 14:45:02.984294:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24276 | 2025-09-27 14:45:02.960352:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24270 | 2025-09-27 14:45:02.947634:  Warn: Inverter control auto restart trigger: REST read failure command []
24269 | 2025-09-27 14:45:02.947541:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24268 | 2025-09-27 14:45:02.934233:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24261 | 2025-09-27 14:45:02.856714:  Warn: Historical day 7 has 10 minutes of gap in the data, filled from 13.55 kWh to make new average 13.65 kWh (percent 99%)
24204 | 2025-09-27 14:40:18.999453:  Completed run status Charging, Hold for car with Errors reported (check log)
24174 | 2025-09-27 14:40:18.851505:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24173 | 2025-09-27 14:40:18.838102:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24167 | 2025-09-27 14:40:18.827478:  Warn: Inverter control auto restart trigger: REST read failure command []
24166 | 2025-09-27 14:40:18.827405:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
24165 | 2025-09-27 14:40:18.815067:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
23940 | 2025-09-27 14:40:02.961434:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
23939 | 2025-09-27 14:40:02.950476:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
23933 | 2025-09-27 14:40:02.940785:  Warn: Inverter control auto restart trigger: REST read failure command []
23932 | 2025-09-27 14:40:02.940703:  Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
23931 | 2025-09-27 14:40:02.929380:  Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
23924 | 2025-09-27 14:40:02.824876:  Warn: Historical day 7 has 10 minutes of gap in the data, filled from 13.62 kWh to make new average 13.71 kWh (percent 99%)
```

GivTCP log:

```
2025-09-27 14:40:02,924 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518f032c0>)
2025-09-27 14:40:02,928 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518c9b380>)
2025-09-27 14:40:02,944 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518ccd140>)
2025-09-27 14:40:02,949 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518cff180>)
2025-09-27 14:40:18,806 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5519063040>)
2025-09-27 14:40:18,813 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f551902ef00>)
2025-09-27 14:40:18,831 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f551907cb00>)
2025-09-27 14:40:18,836 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518eae6c0>)
2025-09-27 14:45:02,926 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518ba2e00>)
[... cut ...]
2025-09-27 14:45:21,579 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f5518f53640>)
2025-09-27 14:45:21,585 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f551902d840>)
2025-09-27 14:45:22,035 - GivTCP - write       -  [INFO    ] - Setting Charge Target 100 was a success
```

gcoan, 2025-09-29:
this is not a predbat error, it is a givtcp problem that it can't decode what the inverter is returning.

There are errors logged on givtcp for this.

I cant use REST as at all on my inverters, it continually fails

closed 2025-10-19 by gcoan