springfall2008/batpred #2658: Not using battery until the last minute, resulting in excess grid import

s0129, 2025-09-14:
**Describe the bug**
Predbat appears to be holding off on using my battery until the end of the day so it ends up on 10% just before the cheap rate. The battery has been full all day but is has been on FrzChrg running from the grid most of the day. This is fine if the load prediction is correct, but it is incorrect today as we are using a lot less electricity so we could have run on battery all day. I expect we will end the day and reach the next cheap period with 50% battery remaining.

**Expected behaviour**
I would expect Predbat to use the battery first as soon as I hit an expensive rate and then switch to grid use if the battery runs out, therefore if the load prediction is incorrect it doesn't matter.

v8.24.2

**Environment details**
- Fox ESS 
- Standard HASS install

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]

[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/22320233/predbat.log)

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/22320238/predbat_debug.yaml.txt)

s0129, 2025-09-16:
Looking at the plan for tomorrow it is trying to do a similar thing, estimating high load so running from the grid instead of using the battery. 

[image]

gcoan, 2025-09-16:
You currently have days_previous and days_previous weight set to use same day last week and the week before as your load prediction:

```
  days_previous:
  - 14
  - 7
  days_previous_weight:
  - 0.5
  - 1
```

So a week ago your load must have been quite high.

You can override the load prediction with select.predbat_manual_load_adjust https://springfall2008.github.io/batpred/customisation/#manual-control

gcoan, 2025-09-22:
It is a bit strange that Predbat goes into freeze and hold charge as the rates go up. Mine doesn't do that!

I can't think of any immediately obvious reason why it would do that, you have high export rates occurring at the same time so maybe its prioritising export, but your PV is dropping right off so that's unlikely.
I can see your SoC is dropping off to almost empty, so whether predbat figures you will run out of battery anyway and so let you grid import for a bit as its the same rate?  Might be worth looking at the predbat plan optimisation switch settings to see if any of these makes a difference.

You can turn charge freeze off as an option but that's a bit of a hammer solution.  As I said it looks like something is misconfigured that is causing the plan to be like this.

Maybe save your current predbat settings (use the select.predbat_save_restore) then go through and set all the configuration settings to the default values to see with a vanilla config how the plan looks.

s0129, 2025-10-29:
@springfall2008 just wondered if you had any advice on this issue, I recently switched to a dynamic tariff (agile) and the workaround to set `predbat_rate_low_threshold` to 0.6 isn't working well. Have found I have to set `num_cars `to 0 to get a sensible looking plan.

s0129, 2025-12-15:
If I disable freeze charge the plan makes even less sense. Seems to be ignoring the cost of charging/discharging the battery?

[image]

gcoan, 2025-12-15:
> Had this a few times, always seems to be after I have had fairly high load for a while when my battery is not large enough to get through the whole day.
> Any idea what I can try to debug it?

If predbat is importing when it doesn't make sense to do so, it's usually because there's a constraint in the config that is causing that.

You've got the force export for Axle in the morning which is lowering the SoC, and then as you say your battery isn't big enough to last you through the rest of the day.

Looking at your debug file you have best_soc_keep set to 2.0.  This could well be the cause of the odd behaviour, you are telling Predbat to try to keep 2kWh in the battery, so it's planning charging to do that.
Clear the best_soc_keep out (set to 0) and see what that does to the plan.

I don't run with any of the best_soc's set, I let Predbat use the entire battery range and if there is a power cut and I don't have enough, then so be it.

chalfontchubby, 2026-09-18:
This issue has drifted a bit after being described as fixed. 

Going to close it. Please open another issue for specific problems if you need to.

closed 2026-09-18 by chalfontchubby