springfall2008/batpred #2668: octopus rates throwing error (possibly due to missing attributes in sensor)

ragg987, 2025-09-20:
**Describe the bug**
Hi, running predbat for the first time today. I am using the Octopus integration in HA and on startup I am getting some errors (note obfuscated Octopus account info)
```
2025-09-20 19:00:07.002330: Warn: No Octopus data in sensor sensor.electricity_NNN_NNNN_previous_rate attribute 'all_rates'
2025-09-20 19:00:07.002423: Warn: No Octopus data in sensor event.electricity_NNN_NNNN_current_day_rates attribute 'all_rates' / 'rates' / 'raw_today' / 'prices'
```
The add-on fails to work and the WebUI shows 
Status | Error: Exception raised

**Expected behaviour**
Add-on to work

**Predbat version**

 Add-on version: 1.2.9
 You are running the latest version of this add-on.

**Environment details**

 My Energi Libbi battery and inverter
 System: Home Assistant OS 16.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.9.1
 Home Assistant Supervisor: 2025.09.0
 Octopus Version 16.3.1
 MyEnergi Version 0.0.30

**Screenshots**
Here are the attributes that HA reports for the octopus current day rates, as an example.
```
event_types: octopus_energy_electricity_current_day_rates
event_type: octopus_energy_electricity_current_day_rates
rates: 
- start: '2025-09-20T00:00:00+01:00'
  end: '2025-09-20T00:30:00+01:00'
  value_inc_vat: 0.070003
  is_capped: false
- start: '2025-09-20T00:30:00+01:00'
  end: '2025-09-20T01:00:00+01:00'
  value_inc_vat: 0.070003
  is_capped: false
- start: '2025-09-20T01:00:00+01:00'
  end: '2025-09-20T01:30:00+01:00'
  value_inc_vat: 0.070003
  is_capped: false

<truncated>
- 
- start: '2025-09-20T23:30:00+01:00'
  end: '2025-09-21T00:00:00+01:00'
  value_inc_vat: 0.070003
  is_capped: false

min_rate: 0.070003
max_rate: 0.289586
average_rate: 0.21181702
mpan: XXX
serial_number: XXX
tariff_code: E-1R-INTELLI-VAR-24-10-29-A
friendly_name: Current Day Rates Electricity (XXX)
intelligent_dispatches_updated: true
```

**Log file**
[predbat.log.txt](https://github.com/user-attachments/files/22444066/predbat.log.txt)

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/22444067/predbat_debug.yaml.txt)

ragg987, 2025-09-20:
To add I used the Libbi yaml kidly shared by @MalcolmSpencer as a starter (thanks!)
https://github.com/springfall2008/batpred/issues/1224#issuecomment-2543886465

gcoan, 2025-09-20:
Changes are you have not followed the setup procedure fully if predbat is not finding the octopus rates.

In particular enabling the Octopus events.  Predbat uses the sensor name to find the event and its the event where the rates are actually read from

gcoan, 2025-09-20:
Can we check the names of the events and the sensors.

Its suspicious in your log that the sensor names are not the same:

```
2025-09-20 19:12:28.734904: Regular expression argument metric_octopus_import matched ^(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)$ with sensor.electricity_NNN_NNNN_current_rate
2025-09-20 19:12:28.735216: Regular expression argument metric_octopus_export matched ^(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_export_current_rate)$ with sensor.octopus_energy_electricity_NNN_NNNN_export_current_rate
2025-09-20 19:12:28.735511: Regular expression argument metric_standing_charge matched ^(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_standing_charge)$ with sensor.electricity_NNN_NNNN_current_standing_charge
```

import has matched sensor.electricity_xxxx whereas the others have sensor.octopus_electricity_xxxx - I think if the sensor and event names don't match it doesn't work

have you renamed the import sensor?

I've seen something like this before on another sensor and there is a warning in the documentation

> NOTE: Predbat automatically joining the saving session relies upon the event being enabled and Predbat being able to find the saving session event as 'event.octopus_energy_ACCOUNT_ID_octoplus_saving_session_events' - there have been occasions with some Octopus Integration installations where the event name is different. If this is the case then the event must be renamed to the correct format that Predbat can function correctly.

both my import and export sensors have the octopus_energy prefix as do my events

and the current_day_rates event has exactly the same attributes as you have so that's not the problem

gcoan, 2025-09-20:
Glad it's sorted.

I have a suspicion that older versions of the Octopus integration didn't name the sensors the same way.  Maybe if you added an export tariff later on it gained the new sensor name?

The sensors are definitely prefixed octopus_energy now.

I'll update the documentation to make it clearer that the event and sensor name must have the same prefix.

ragg987, 2025-09-20:
Yes that is possible. I installed the Octopus add-on over 2 years ago, I was with Agile then. About a year ago I both switched to IOG and added Export tariff.

gcoan, 2025-09-21:
Documentation updated

closed 2025-09-21 by gcoan