springfall2008/batpred #2670: Huawei apps.yaml

ward0, 2025-09-21:
Errors:
<html>
<body>

6474 | 2025-09-21 20:12:05.305395: Warn: Validation of apps.yaml found configuration item 'discharge_stop_service' element huawei_solar/stop_forcible_charge is not an dict
-- | --
6473 | 2025-09-21 20:12:05.305344: Warn: Validation of apps.yaml found configuration item 'discharge_start_service' element huawei_solar/forcible_discharge_soc is not an dict
6472 | 2025-09-21 20:12:05.305293: Warn: Validation of apps.yaml found configuration item 'charge_stop_service' element huawei_solar/stop_forcible_charge is not an dict
6471 | 2025-09-21 20:12:05.305228: Warn: Validation of apps.yaml found configuration item 'charge_start_service' element huawei_solar/forcible_charge_soc is not an dict
6470 | 2025-09-21 20:12:05.304679: Warn: Validation of apps.yaml found configuration item 'discharge_end_time' element 23:00:00 is not a valid entity_id (must contain a dot)
6469 | 2025-09-21 20:12:05.304618: Warn: Validation of apps.yaml found configuration item 'discharge_start_time' element 07:00:00 is not a valid entity_id (must contain a dot)
6468 | 2025-09-21 20:12:05.304556: Warn: Validation of apps.yaml found configuration item 'scheduled_discharge_enable' element False is not a valid entity_id (must be a string)
6467 | 2025-09-21 20:12:05.304493: Warn: Validation of apps.yaml found configuration item 'scheduled_charge_enable' element False is not a valid entity_id (must be a string)
6466 | 2025-09-21 20:12:05.304383: Warn: Validation of apps.yaml found configuration item 'charge_end_time' element 06:00:00 is not a valid entity_id (must contain a dot)
6465 | 2025-09-21 20:12:05.304295: Warn: Validation of apps.yaml found configuration item 'charge_start_time' element 00:00:00 is not a valid entity_id (must contain a dot)

</body>
</html>

Apps.yaml configuration:

  # Services to charge/discharge
  charge_start_service: huawei_solar/forcible_charge_soc
  charge_stop_service: huawei_solar/stop_forcible_charge
  discharge_start_service: huawei_solar/forcible_discharge_soc
  discharge_stop_service: huawei_solar/stop_forcible_charge

  scheduled_charge_enable:
    - off
  scheduled_discharge_enable:
    - off
  charge_start_time:
    - "00:00:00"
  charge_end_time:
    - "06:00:00"
  discharge_start_time:
    - "07:00:00"
  discharge_end_time:
    - "23:00:00"

  battery_min_soc:
    - 5

I think the template yaml is very out of date: https://github.com/springfall2008/batpred/blob/main/templates/huawei.yaml

emilio-valdes, 2025-11-12:
> Thank you @gcoan<https://github.com/gcoan> I’ll ignore the warning, I have moved predbat to control charge and discharge, and I’ll monitor it closely. I’ll let you know how it goes. If it does not work, then I’ll try the other solution you suggested
>
> Be good to get this resolved to stop unnecessary warnings even though they can safely be ignored, so if you can try the replacement service calls, and correct/confirm them, that would be good.
>
> Good luck with Predbat Control Charge and Discharge !
>
> Cheers

I just tried. Same warning.

49718
2025-11-12 14:14:26.818187: Error: Validation of apps.yaml found 7 configuration errors
49717
2025-11-12 14:14:26.817969: Warn: Validation of apps.yaml found configuration item 'discharge_stop_service' element huawei_solar.stop_forcible_charge is not an dict
49716
2025-11-12 14:14:26.817478: Warn: Validation of apps.yaml found configuration item 'discharge_start_service' element huawei_solar.forcible_discharge_soc is not an dict
49715
2025-11-12 14:14:26.817071: Warn: Validation of apps.yaml found configuration item 'charge_stop_service' element huawei_solar.stop_forcible_charge is not an dict
49714
2025-11-12 14:14:26.816601: Warn: Validation of apps.yaml found configuration item 'charge_start_service' element huawei_solar.forcible_charge_soc is not an dict
49713
2025-11-12 14:14:26.815480: Warn: Validation of apps.yaml found configuration item 'scheduled_discharge_enable' element off is not a valid entity_id (must contain a dot)
49712
2025-11-12 14:14:26.815244: Warn: Validation of apps.yaml found configuration item 'scheduled_charge_enable' element off is not a valid entity_id (must contain a dot)
49711
2025-11-12 14:14:26.814959: Warn: Validation of apps.yaml found configuration item 'charge_end_time' element input_datetime.tarifa_valle_end" returned value None
49706
2025-11-12 14:14:23.385413: Warn: Regular expression argument: pv_forecast_d4 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_4|d4)), now will disable
49705
2025-11-12 14:14:23.380530: Warn: Regular expression argument: pv_forecast_d3 unable to match re:(sensor.(solcast_|)(pv_forecast_|)forecast_(day_3|d3)), now will disable

gcoan, 2025-11-16:
@ward0 @emilio-valdes  I have updated the template Huawei apps.yaml to add the grid power sensor.  I have not added the inverts, but per the above, think they might be required

Please check you are happy with the changes: https://github.com/gcoan/batpred/commit/c13c2fd1309840c65baf2f939585fe03ee3fec4d

gcoan, 2025-11-26:
No news is good news so closing this issue.  Template and documentation changes will be in my next PR

gcoan, 2025-11-26:
The recommended apps.yaml entries are:

```
  # Services to charge/discharge
  charge_start_service: 
    - service: script.charge_battery_from_grid
  charge_stop_service:
    - service: huawei_solar.stop_forcible_charge
      device_id: "{device_id}"
  discharge_start_service:
    - service: huawei_solar.forcible_discharge_soc
      device_id: "{device_id}"
  discharge_stop_service:
    - service: huawei_solar.stop_forcible_charge
      device_id: "{device_id}"
```

and per the latest setup guide:

Huawei inverters can charge the battery from DC solar and discharge at one power level (e.g. 5kWh), but have a lower limit (e.g. 3kWh) for AC charging. At present Predbat doesn't have the ability to model separate DC and AC charging limits, so battery_rate_max is set to the upper limit in watts (e.g. 5000) in the template apps.yaml to ensure that Predbat doesn't limit DC charging and discharging.
Create the following automation script which Predbat will call to start grid battery charging when required, customise the power rating to charge at, and enter your own Huawei device id:

```
alias: Charge battery from grid
description: Start charging Huawei battery from grid at AC charge limit
sequence:
  - action: huawei_solar.forcible_charge_soc
    data:
      target_soc: 100
      power: "3000"
      device_id: XXXXXXXX
```

On issue #2982 the other person with a Huawei inverter has got this working ok.  He says it needed a script for discharging as well, but the above should work for charging

gcoan, 2025-11-28:
Looking at your logfile I can see you have days_previous set to just '7', i.e. it is only taking the load information from a single day, a week ago.  This will work, but its not great because if you have got a problem with that history on that day or it was a particularly high or low consumption day then it will throw your plan off.

It is better if you can to average multiple days of history, e.g. I have:

```
days_previous:
  - 2
  - 3
  - 4
  - 5
  - 6
  - 7
  - 8
```

so I average over a whole week

What is in your load sensor 

One other thing I noticed:

export limit 5999940.0 kW

it is defined in apps.yaml in watts, looks like you have an absolutely massive value set for it?