springfall2008/batpred #2681: 8.25.1 Python issue

ASTPlatinum, 2025-09-22:
Traceback (most recent call last): File "/config/hass.py", line 5, in <module> import predbat File "/config/predbat.py", line 78, in <module> from components import Components File "/config/components.py", line 13, in <module> from ohme import OhmeAPI File "/config/ohme.py", line 13, in <module> from typing import Any, Optional, Self, Mapping ImportError: cannot import name 'Self' from 'typing' (/usr/lib/python3.10/typing.py)

Home Assistant OS 16.2 (Supervisor 2025.09.0) running on Proxmox
Predbat version: 8.25+

Reverted back to Predbat v8.24.2

2ZZ, 2025-09-22:
> could one of you attach the full logfile please

```
2025-09-22 21:50:25.314914 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2025-09-22 21:50:25.315073 INFO AppDaemon: Python version is 3.10.11
2025-09-22 21:50:25.315137 INFO AppDaemon: Configuration read from: /conf/appdaemon.yaml
2025-09-22 21:50:25.315198 INFO AppDaemon: Added log: AppDaemon
2025-09-22 21:50:25.315264 INFO AppDaemon: Added log: Error
2025-09-22 21:50:25.315314 INFO AppDaemon: Added log: Access
2025-09-22 21:50:25.315373 INFO AppDaemon: Added log: Diag
2025-09-22 21:50:25.345248 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2025-09-22 21:50:25.373630 INFO HASS: HASS Plugin Initializing
2025-09-22 21:50:25.373731 INFO HASS: HASS Plugin initialization complete
2025-09-22 21:50:25.373943 INFO AppDaemon: Initializing HTTP
2025-09-22 21:50:25.374099 INFO AppDaemon: Using 'ws' for event stream
2025-09-22 21:50:25.375780 INFO AppDaemon: Starting API
2025-09-22 21:50:25.377734 INFO AppDaemon: Starting Admin Interface
2025-09-22 21:50:25.377878 INFO AppDaemon: Starting Dashboards
2025-09-22 21:50:25.391267 INFO HASS: Connected to Home Assistant 2025.8.3
2025-09-22 21:50:25.444913 INFO HASS: Evaluating startup conditions
2025-09-22 21:50:25.451851 INFO HASS: Startup condition met: hass state=RUNNING
2025-09-22 21:50:25.451991 INFO HASS: All startup conditions met
2025-09-22 21:50:25.470174 INFO AppDaemon: Got initial state from namespace default
2025-09-22 21:50:25.512045 WARNING AppDaemon: File '/conf/apps/predbat/config/apps.yaml' duplicate app: pred_bat - ignoring
2025-09-22 21:50:25.512520 INFO AppDaemon: App 'pred_bat' added
2025-09-22 21:50:25.512839 INFO AppDaemon: Found 1 active apps
2025-09-22 21:50:25.512935 INFO AppDaemon: Found 0 inactive apps
2025-09-22 21:50:25.513022 INFO AppDaemon: Found 0 global libraries
2025-09-22 21:50:25.513117 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2025-09-22 21:50:25.513455 INFO AppDaemon: Running on port 5050
2025-09-22 21:50:26.516258 INFO AppDaemon: Scheduler running in realtime
2025-09-22 21:50:26.518414 INFO AppDaemon: Adding /conf/apps to module import path
2025-09-22 21:50:26.518822 INFO AppDaemon: Adding /conf/apps/predbat to module import path
2025-09-22 21:50:26.519426 INFO AppDaemon: Adding /conf/apps/predbat/config to module import path
2025-09-22 21:50:26.532582 WARNING AppDaemon: No app description found for: /conf/apps/hello.py - ignoring
2025-09-22 21:50:26.533025 WARNING AppDaemon: No app description found for: /conf/apps/predbat/config.py - ignoring
2025-09-22 21:50:26.533451 WARNING AppDaemon: No app description found for: /conf/apps/predbat/db_engine.py - ignoring
2025-09-22 21:50:26.533868 WARNING AppDaemon: No app description found for: /conf/apps/predbat/prediction.py - ignoring
2025-09-22 21:50:26.534293 WARNING AppDaemon: No app description found for: /conf/apps/predbat/plan.py - ignoring
2025-09-22 21:50:26.534658 WARNING AppDaemon: No app description found for: /conf/apps/predbat/userinterface.py - ignoring
2025-09-22 21:50:26.535018 WARNING AppDaemon: No app description found for: /conf/apps/predbat/fetch.py - ignoring
2025-09-22 21:50:26.535371 WARNING AppDaemon: No app description found for: /conf/apps/predbat/execute.py - ignoring
2025-09-22 21:50:26.535704 WARNING AppDaemon: No app description found for: /conf/apps/predbat/download.py - ignoring
2025-09-22 21:50:26.536038 WARNING AppDaemon: No app description found for: /conf/apps/predbat/web.py - ignoring
2025-09-22 21:50:26.536404 WARNING AppDaemon: No app description found for: /conf/apps/predbat/unit_test.py - ignoring
2025-09-22 21:50:26.536737 WARNING AppDaemon: No app description found for: /conf/apps/predbat/gecloud.py - ignoring
2025-09-22 21:50:26.537067 WARNING AppDaemon: No app description found for: /conf/apps/predbat/octopus.py - ignoring
2025-09-22 21:50:26.537426 WARNING AppDaemon: No app description found for: /conf/apps/predbat/futurerate.py - ignoring
2025-09-22 21:50:26.537820 WARNING AppDaemon: No app description found for: /conf/apps/predbat/utils.py - ignoring
2025-09-22 21:50:26.538183 WARNING AppDaemon: No app description found for: /conf/apps/predbat/alertfeed.py - ignoring
2025-09-22 21:50:26.538543 WARNING AppDaemon: No app description found for: /conf/apps/predbat/predheat.py - ignoring
2025-09-22 21:50:26.538929 WARNING AppDaemon: No app description found for: /conf/apps/predbat/energydataservice.py - ignoring
2025-09-22 21:50:26.539312 WARNING AppDaemon: No app description found for: /conf/apps/predbat/ohme.py - ignoring
2025-09-22 21:50:26.539680 WARNING AppDaemon: No app description found for: /conf/apps/predbat/compare.py - ignoring
2025-09-22 21:50:26.540117 WARNING AppDaemon: No app description found for: /conf/apps/predbat/solcast.py - ignoring
2025-09-22 21:50:26.540510 WARNING AppDaemon: No app description found for: /conf/apps/predbat/web_helper.py - ignoring
2025-09-22 21:50:26.540910 WARNING AppDaemon: No app description found for: /conf/apps/predbat/fox.py - ignoring
2025-09-22 21:50:26.541389 WARNING AppDaemon: No app description found for: /conf/apps/predbat/plugin_system.py - ignoring
2025-09-22 21:50:26.541846 WARNING AppDaemon: No app description found for: /conf/apps/predbat/inverter.py - ignoring
2025-09-22 21:50:26.542233 WARNING AppDaemon: No app description found for: /conf/apps/predbat/db_manager.py - ignoring
[... cut ...] iohttp', 'octopus', 're', 'yaml', 'web', 'shutil', 'web_helper', 'urllib.parse', 'html', 'ohme', 'dataclasses', 'enum', 'fox', 'hashlib', 'random', 'argparse', 'gecloud', 'userinterface', 'solcast', 'sys', 'alertfeed', 'xml.etree.ElementTree', 'predheat', 'output', 'math', 'prediction', 'gc', 'download', 'plan', 'multiprocessing', 'compare'] already, but fetch depends on {'futurerate', 'utils', 'config', 'datetime'}
2025-09-22 21:51:08.855619 INFO AppDaemon: Previous message repeated 1 times
```

gcoan, 2025-09-22:
> Container works for me, are you referring to this one? https://github.com/nipar4/predbat_addon

yes thats the docker container that's supported.

@ASTPlatinum's use of appdaemon may be unrelated if you are both seeing the same issue, but I thought it might be the cause as appdaemon has been pretty much retired for predbat

ASTPlatinum, 2025-09-27:
Still an issue with 8.25.3

Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.25.3
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 78, in <module>
    from components import Components
  File "/config/components.py", line 13, in <module>
    from ohme import OhmeAPI
  File "/config/ohme.py", line 13, in <module>
    from typing import Any, Optional, Self, Mapping
ImportError: cannot import name 'Self' from 'typing' (/usr/lib/python3.10/typing.py)

ASTPlatinum, 2025-09-27:
@springfall2008 I've reinstalled HA on proxmox pulling the lastest version, restored a backup, then updated predbat and the same issue remains.
Rebuilt predbat - same issue. Removed predbat and re-installed - Same issue.

Now used good old ChatGPT and it maybe chatting utter BS but it says The traceback shows /usr/lib/python3.10/typing.py → that’s the add-on’s Python, not HA Core’s.