springfall2008/batpred #2686: predheat error:

ragg987, 2025-09-24:
**Describe the bug**
predheat will throw an error - see extract from log below. This error is currently intermittent - e.g. my setup ran overnight fine then this morning the error showed up even though I had made no changes. However I have also been able to force the error by changing:
heat_loss_watts
heat_loss_degrees

If these parameters are higher than currently set - e.g. HLW = 250 then an error occurs straightaway.

Also note if I switch off the predheat toggle then my predbat is working and stable.

**Expected behaviour**
No error.

**Predbat version**
8.25.1

**Environment details**
 Add-on version: 1.2.9
 System: Home Assistant OS 16.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.9.1
 Home Assistant Supervisor: 2025.09.0

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
```
2025-09-24 07:43:08.090724: Info: Start socket for url http://supervisor/core/api/websocket
Web interface started
2025-09-24 07:43:08.099174: Info: Web Socket active
Watching ['./inverter.py', './ohme.py', './utils.py', './userinterface.py', './alertfeed.py', './octopus.py', './output.py', './energydataservice.py', './predheat.py', './apps.yaml', './futurerate.py', './hass.py', './config.py', './predbat.py', './unit_test.py', './solcast.py', './components.py', './ha.py', './compare.py', './web.py', './fox.py', './web_helper.py', './execute.py', './db_manager.py', './plugin_system.py', './gecloud.py', './fetch.py', './plan.py', './db_engine.py', './download.py', './prediction.py'] for changes
2025-09-24 07:43:37.999166: Info: record_status Demand
2025-09-24 07:43:38.556449: ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType'
2025-09-24 07:43:38.568188: Info: record_status ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType'
2025-09-24 07:43:38.568271: Error: '<' not supported between instances of 'float' and 'NoneType'
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predheat.py", line 642, in update_time_loop
    raise e
  File "/config/predheat.py", line 638, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predheat.py", line 586, in update_pred
    next_volume_temp, predict_minute = self.run_simulation(self.next_volume_temp, self.heating_active)
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/predheat.py", line 382, in run_simulation
    if volume_temp < flow_temp:
       ^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '<' not supported between instances of 'float' and 'NoneType'
```

[apps.yaml](https://github.com/user-attachments/files/22509222/apps.yaml)

**Predbat debug yaml file**

symonchester, 2025-10-02:
I also see the same error and I have predheat active

springfall2008, 2025-10-06:
This implies flow_temp is not set correctly

ragg987, 2025-10-06:
I'm using the flow sensor from my ASHP, which flows at around 28 to 32C. Extract from apps.yaml
```
    # Current flow temperature setting
    flow_temp: sensor.yutaki_1093_temp_water_out
    flow_difference_target: 4
```

[image]

symonchester, 2026-01-04:
The error seems to have returned again, yesterday at 7am
It cleared on its own last time but appears to have returned 

ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType' (unhealthy)

gcoan, 2026-01-04:
> The error seems to have returned again, yesterday at 7am It cleared on its own last time but appears to have returned
> 
> ERROR: Exception raised '<' not supported between instances of 'float' and 'NoneType' (unhealthy)

it could probably be trapped with a better error message in Predbat, but this is your ASHP flow temperature briefly not being able to get data and returning a 'none' value which is what predbat errors over.

I think if you look at the sensor history at the time that predbat errored you should see it.

symonchester, 2026-01-04:
Thanks for getting back to me...
Looking at the history (6am yesteday) it seems the ASHP Flow temp was reporting data as expected.

[image]

symonchester, 2026-01-04:
Its happened twice now, once back in October and yesterday/today..

Files attached thanks

[predbat_debug.yaml](https://github.com/user-attachments/files/24423878/predbat_debug.yaml)
[predbat.log](https://github.com/user-attachments/files/24423877/predbat.log)

gcoan, 2026-01-11:
180 days is more than enough entity history. Strange that predheat reports finding 1 days of history then

I would think your HA database is enormous.  I had my history at 30 days and the database was over 2Gb, I'm now down to 11 days and for a lot of entities I purge them down to just 1 day to keep the database size manageable.  

Anyway, must be something else.

bencope, 2026-09-03:
I believe this is due to the weather compensation.

From your yaml you have this set
```yaml
    weather_compensation:
      -2: 47.0
      16: 23.0
```

right before the IF that is causing the issue is this

```python
                out_temp = int(external_temp + 0.5)
                out_temp = min(max(out_temp, -20), 20)
                heat_to = target_temp

                if self.weather_compensation_enabled:
                    flow_temp = self.weather_compensation.get(out_temp)
                else:
                    flow_temp = self.flow_temp

                if volume_temp < flow_temp:
```

because you have weather comp enabled the flow_temp is a function of out_temp which is constrained between -20 and 20, however your config's upper bound is 16 which means flow_temp is set to NoneType when external_temp + 0.5 > 16

In the main docs https://springfall2008.github.io/batpred/predheat/#weather-compensation the example is like this which I guess is not by accident but would be nice if it was explicitly stated in the docs
```yaml
  weather_compensation:
    -20: 45.0
    -3: 45.0
    15: 25.0
    20: 25.0
```

Anyway, I was suffering with this for months (since the spring...when it got warmer) but since I've adjusted my config to cover -20 to 20 I've not seen this issue.

Hope this helps!

gcoan, 2026-09-03:
@bencope thanks for your input

just so I understand what is wrong, you're saying that the weather compensation curve needs to always have high and low values of -20 and +20, otherwise you'll get this non type error when Predheat tries to convert outside temperature to flow temperature?

I can certainly update the documentation to make this explicit. 

thanks

bencope, 2026-09-03:
Yes that’s my understanding from the code. Could have higher or lower bounds but it always needs to cover that range.

gcoan, 2026-09-03:
Thanks I have added:

**IMPORTANT**: The weather compensation curve must have a low value of -20 degrees C and a high value of 20 degrees C.  You can have whatever other points you want between the upper and lower bounds and Predheat will fill in the gaps, but if you don't have -20 and 20 Predheat will error when it tries to calculate the weather-adjusted flow temperature.

It'll be part of my PR that's pending approval from Trefor.

gcoan, 2026-09-03:
@ragg987 can you can test whether this fixes your problem.  The issue has been open for 9 months or so and we are trying to clear some of the backlog.  We'll close it in a week or so if its not reported as still being a problem

ragg987, 2026-09-04:
I've set the -20 and 20 parameters and predbat has not thrown this error yet (8hrs of running), so nice find. I had stripped out all my predheat charts etc so might have to put them back in once the heating season starts.

gcoan, 2026-09-04:
> I've set the -20 and 20 parameters and predbat has not thrown this error yet (8hrs of running), so nice find. I had stripped out all my predheat charts etc so might have to put them back in once the heating season starts.

excellent, fingers crossed that this was the fix you needed

chalfontchubby, 2026-09-15:
Could you confirm this is fixed please - in which case please close it.

closed 2026-09-15 by chalfontchubby