springfall2008/batpred #2687: Import and export block for energy rates

gcoan, 2025-09-24:
**Is your feature request related to a problem? Please describe.**
Predbat does a brilliant job of optimising the battery use, but sometimes it tries too hard, planning repeated charge and discharge cycles overnight (e.g. on iGo) to maximise the income, but at the expense of extra battery cycles.

Whilst you can give predbat manual overrides for a specific day to change the plan, it would be useful (and simpler for users) to be able to explicitly direct predbat for periods that you don't want import or export.

**Describe the solution you'd like**
Something like the ability to configure a (date), start time and end time when import or export is to be prevented in apps.yaml

e.g.

[image]

Predbat will then take account of these in the plan.

**Describe alternatives you've considered**
Automations to set a series of force charge/demand/export commands at the required times.  Possible but messy.

**Additional context**
Strangely Chat GPT thinks predbat already has this feature.  If you ask it how to prevent predbat exporting overnight it suggests using export_block!

ragg987, 2025-10-06:
> I make use of "switch.predbat_combine_charge_slots" = true to keep the battery doing one during the cheap IOG overnight slot.

Thanks for that tip - I had read it twice in the docs but did not get it :grimacing:. Now tried that suggestion, and combined with a small price over-ride I have whay I need. i.e. battery sits with a small charge for the half the cheap window then ramps up to maximum charge in the second half of that window.

[image]

reydelamirienda, 2026-03-19:
I'm also on IOG with Agile export. Predbat goes bananas for the massive spreads in the evening (7p import, 20p+ export) and takes the battery on a rollercoaster ride. It's only going to get worse from April when IOG import goes down to 3.5p.

In my case I've found a simple solution: override export rate to 0p:

```
rates_export_override:
    - start: '23:30:00'
      end: '05:30:00'
      rate: 0
```

chalfontchubby, 2026-08-21:
We have metric_batter_cycle which models the wear and tear cost of cycling a battery - rather than hard coding a "don't charge at this time" that deters cycling when it's not worth it for small gain. 

Check the docs, but a value of a few pence helps me avoid microcycling - while if I _could_ buy at 3.5p and sell at 20p+ I'd happily take that cycling.

gcoan, 2026-08-21:
I think metric battery cycle cost has been tried, but one of the EV tariff owners would need to confirm.

Overnight with a cheap rate EV slot, the use case is wanting the battery to be charged overnight (and then all solar exported the next day), but to reduce wear and tear on the battery (esp GivEnergy given no warranty any more), the ask is that repeated exports and re-imports are not done.

The proposed import/export block is to define periods where you don't want predbat to do certain actions regardless of the financial benefit of doing so.

chalfontchubby, 2026-08-21:
metric_battery_cycle isn't mentioned anywhere in the issue, and works to prevent cycling for me in this case. There may still be a case for not making financially beneficial charging decisions even taking into account the wear on the battery, but this should be tried first - I'm on about 3p/kWh with good results. 

I'm not going to prioritise for now - it can stay on the backlog until someone picks it up.

PianSom, 2026-08-21:
Following up on the above - I think this would be a great enhancement.

Like others, I have significant issues with Predbat behaving unpredictably/irrationally. AIO1x2+GW system. I cycle batteries overnight to make income while my unsupported system remains alive - so I do not want to mess with battery cycling costs. The behaviour of combine_charge and combine_dischage slots is what leads to the unpredictable behaviour. (I have them both OFF currently, though have experimented extensively). I am addressing this by having HA introduce a manual discharge 2-3.30am, and have Metric Min Export Improvement set to 24p (ie 2kWh).

chalfontchubby, 2026-08-21:
Please file issues showing “behaves irrationally” with logs we can action. I don’t follow why battery cycling costs is the wrong fix for the issues reported here - it’s cycling leading to wear on the battery for little gain.  I’m not saying there isn’t a case for this and anyone can pick it up to implement. I’m just think from the discussion here it would fix a symptom and not the source of the problem.

peter1rhodes, 2026-08-21:
One reason that in my view, the `metric_battery_cycle` approach may not adequately fix the issue is this: I can import at 3.5p and export at 12p. This is enough of a gap to justify discharging the battery daily in my view, so in general at these rates, I want the battery to fill overnight (regardless of solar expectations) and discharge at some point during the day. Thus, I don't want to set my cycle cost to be so high that buying at 3.5 and selling at 12 does not happen at all. I simply don't think it's helping the grid (and might annoy my provider) if I do so all night during the cheap import rate, constantly pumping up and down. I can justify the first behaviour because I'm (hopefully) able to help the grid during high demand in the day, but the other is clearly not good behaviour. However, as I understand it, `metric_battery_cycle` is not able to make this distinction

chalfontchubby, 2026-08-21:
No - metric_battery_cycle doesn't differentiate between arbitrage at night vs during the day. I'm kinda jealous of a 3.5p rate.

I will consider a fix - unless someone else gets to it first - there is just a long list of issues to look at.

chalfontchubby, 2026-08-29:
Coming back to this with a concrete proposal that I think covers most of what's been asked for here without inventing a new subsystem.

## The timer half already exists

`rates_export_override` already takes `start`, `end`, `date`, **`day_of_week`** (1-7, comma separated) and a `utc` flag, and gets replicated across future days. So a recurring window is already expressible today:

```yaml
rates_export_override:
  - start: "23:30:00"
    end: "05:30:00"
    rate: 0
    day_of_week: "1,2,3,4,5"
```

Predbat won't choose to export for 0p, so this already behaves as a block. It's documented, but scattered - `day_of_week` is in energy-rates.md and the "override the night export rate to zero" advice is a single line in configuration-guide.md, under none of the words anyone would search for. That's most of why this keeps coming up.

## What's actually missing: clamping instead of replacing

The gap is that `rate:` **replaces** the rate rather than bounding it. Two consequences:

- If the real export rate is ever negative in that window, overriding to `0` makes exporting *more* attractive than reality - the opposite of the intent.
- @peter1rhodes' current practice ("manually override my export rate during the cheap period to be the same as the import rate") can't be expressed as a bound, only as a hardcoded replacement.

So I'd propose adding `rate_min` and `rate_max` alongside the existing `rate` and `rate_increment`:

- `rate_max: 0` → `min(existing_rate, 0)`. A real +15p export becomes 0 and is blocked; a genuine -5p stays -5p and stays unattractive.
- `rate_min: 40` → `max(existing_rate, 40)`. Makes grid charging in the window unattractive without pretending import is free.

This is deliberately minimal. The application site in `basic_rates()` is already a two-branch (`rate_increment` vs replace); this makes it a four-way. No optimiser changes, no new scheduling concept, and it reuses all the existing window/date/day-of-week/UTC handling. It's also generally useful beyond this issue - "cap my export rate at X during this window" isn't a single-purpose knob.

An `export_block`/`import_block` name could sit on top as sugar later if the plain syntax still isn't discoverable enough, but I'd rather see whether documenting it properly is sufficient first.

## Two things to be honest about

**It's a strong disincentive, not an absolute bar.** With export capped at 0 there is still one case where Predbat will export: when the battery is full and PV would otherwise be clipped, or when rates are about to go negative. Exporting for nothing beats wasting it, and I'd argue that's the *right* answer for anyone whose motivation here is "doing the right thing" rather than the money - it's not arbitrage, it's not wasting generation, and it helps rather than hurts if the grid is long later in the day. If you want a guarantee that it never exports under any circumstances, this approach won't give you that.

**The import side distorts the displayed plan cost.** House load import is unavoidable, so a floored import rate prices that unavoidable load at the fake rate. Block 23:30-05:30 with `rate_min: 100` and a 0.3 kW baseline and the plan will show roughly £1.74/night of cost that won't actually be billed. The floor only has to beat your peak rate to stop Predbat choosing to grid-charge - `rate_min: 40` does the job with far less distortion than `100`. Worth knowing before picking a number.

## What this doesn't do

It doesn't cover the dynamic version I floated earlier in this thread - blocking export whenever the *import* rate is below a threshold, which would catch IOG's dynamic cheap slots wherever they land rather than only in a fixed window. That's a genuinely different and larger piece of work and I think it should stay a separate discussion.

Does this look like it covers the actual need for people here? If the shape seems right I'll look at putting it together.

antlear, 2026-10-19:
@chalfontchubby  Where did you place the timer and is it working ok?

rates_export_override:
  - start: "23:30:00"
    end: "05:30:00"
    rate: 0
    day_of_week: "1,2,3,4,5"

chalfontchubby, 2026-10-19:
I have not used the timer myself. I don't have overnight micro cycles. Changes here got swamped so far in the back lock of bugfixes and other feature requests.

gcoan, 2026-10-19:
@antlear the rates_export_override feature already exists in predbat, its documented in the energy_rates part of the documentation, you simply need to add the lines to your apps.yaml

@chalfontchubby sorry I overlooked your suggested way to implement this.  It sounds a good approach that avoids significant upheaval to the core predbat planning code. 

In the UK the rate_min isn't actually needed to achieve this as even on Agile export the rate doesn't go negative, it floors at zero.  However I know in Australia and possibly Holland their rates do go negative so would be worth including it for global compatibility.

The way this manifests in cost_today is my only real concern about this proposal and which is why I'd shied away from using rate adjustments to deliver this because you get incorrect cost calculations in Predbat.  Less so for export where a zero should persuade Predbat to not export so it doesn't actually impact the plan costing (notwithstanding export to reduce PV clipping issue you note), but definitely an issue for overridden import rates.
Is there any way to introduce these min/max rates for the planning engine but leave the actual costs in the cost_today calculation?