springfall2008/batpred #2689: My inverter was offline for a while due to wfi issues. Now it is not restarting with an invalid status error

mehtadone, 2025-09-25:
**Describe the bug**
My inverter was offline due to wifi issues. Now the inverter is back online but predbat cannot start properly. I have made no changes to config and was working well before the wifi issues.

The Fox Cloud app is showing data correctly and so is the Fox Modbus integration in HA 

**Expected behaviour**
I expect predbat to start correctly like it did before. 

**Predbat version**

1.29

**Environment details**

- Fox ESS H8
- Home Assistant on HAOS

**Log file**
```
IndexError: list index out of range
2025-09-25 08:20:04.157492: Warn: PV Forecast today adds up to 79.66 kWh but total sensors add up to 23.05 kWh, this is unexpected and hence data maybe misleading (factor 3.5)
2025-09-25 08:20:04.178199: Warn: Empty history passed to minute_data, ignoring (check your settings)...
2025-09-25 08:20:12.814972: Error: Exception raised list index out of range
2025-09-25 08:20:12.817687: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1492, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 683, in update_pred
    self.fetch_inverter_data()
  File "/config/execute.py", line 663, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/inverter.py", line 592, in find_charge_curve
    predbat_status = self.base.minute_data_state(predbat_status_data[0], self.base.max_days_previous, self.base.now_utc, "state", "last_updated")
                                                 ~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2025-09-25 08:20:12.861789: Info: record_status Error: Exception raised list index out of range
2025-09-25 08:20:12.861892: Error: list index out of range
Traceback (most recent call last):
  File "/config/hass.py", line 191, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 1497, in run_time_loop
    raise e
  File "/config/predbat.py", line 1492, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 683, in update_pred
    self.fetch_inverter_data()
  File "/config/execute.py", line 663, in fetch_inverter_data
    curve = inverter.find_charge_curve(discharge=False)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/inverter.py", line 592, in find_charge_curve
    predbat_status = self.base.minute_data_state(predbat_status_data[0], self.base.max_days_previous, self.base.now_utc, "state", "last_updated")
                                                 ~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

mehtadone, 2025-09-25:
Thank you @gcoan Adding the charge curves into the config worked and I have a plan calculated again. 

Will wait for a fix before removing these.

springfall2008, 2026-09-22:
**Automated first-pass triage** — a maintainer will review before any action is taken.

**Classification: bug** — `priority_low` (already fixed upstream; only affects versions older than v8.26.2)

What I investigated:

- Your traceback points at `predbat_status_data[0]` inside `find_charge_curve` (`inverter.py:592` in the version you were running, ~v8.25.x — the "1.29" is the HA add-on version). At that time the guard before the indexing read `if soc_kwh_data and charge_rate_data and charge_rate_data and battery_power_data:` — `charge_rate_data` is listed twice and `predbat_status_data` is never checked, so an empty history list for `predbat.status` passed the guard and the `[0]` raised `IndexError`.
- This is the same crash as #2806 (v8.26.1 on Sofar), which was fixed by PR #2814 (merged 2025-10-23, first released in **v8.26.2**): it adds `required=False` to all four `get_history_wrapper` calls and corrects the guard to `if soc_kwh_data and charge_rate_data and battery_power_data and predbat_status_data:`. On current main (`apps/predbat/inverter.py:1158`) an empty history now simply skips curve detection instead of crashing — no unit test reproduces the exact outage scenario, but the guard change is the fix for the identical traceback.
- The "Empty history passed to minute_data" warning earlier in your log is consistent with HA having no usable history for that window after the inverter outage.

The hardcoded-curves workaround you applied is confirmed working, and the fix shipped in v8.26.2 (Oct 2025). Please update Predbat to a current release and remove the curves from apps.yaml; if the crash still occurs on a current version, post a fresh log here.

chalfontchubby, 2026-10-19:
_Posted by Claude on behalf of @chalfontchubby._

Closing as fixed. The `IndexError` in `find_charge_curve` after the inverter had been offline was fixed by #2814 (released in v8.26.2): the curve search now only runs when the SoC, charge-rate, battery-power and status histories are all present. Please reopen if you see it again on a current release.

closed 2026-10-19 by chalfontchubby