springfall2008/batpred #2690: Web UI Compare page

brickatius, 2025-09-25:
**Describe the bug**
In the latest versions of the Predbat Web UI 'Compare' page the Tariff Comparison - True cost chart does not load. There is a persistent Loading chart (please wait)...... message. 

**Expected behaviour**
You would expect the Tariff Comparison chart to load in the same way as it did in v8.24.1

**Predbat version**
v8.24.2 and above

xxxx

**Environment details**

- LuxPower Inverter with 9.6kWh battery storage
- Standard HAOS 16.2 
- Core 2025.09.0
- Everything else is working fine

[image]

[image]

[predbat-2.log](https://github.com/user-attachments/files/22534411/predbat-2.log)

[predbat_debug-2.yaml](https://github.com/user-attachments/files/22534515/predbat_debug-2.yaml)

brickatius, 2025-09-25:
It works consistently on 8.24.1 but has been patchy on 8.24.2 and I definitely cannot get it to work on 8.25.0 or 8.25.1
I don't think it's a browser issue. I've been looking on my iPhone and macbook (both Safari) and also my Linux machine running Ubuntu (Firefox). I can access HA locally and also via a VPN (Tailscale) on all 3 devices. I've deleted history and cleared the cache on all 3 machines all to no avail. If it works on one, it works on all and vice versa. 
Obviously Predbat has restarted every time I've swapped from one version to another and I've also restarted Home Assistant. I haven't installed anything on HA recently, only done the routine updates. Unfortunately I haven't got any HA backups older than a couple of days so I can't roll back to an earlier version of HA.
The only thing I can think if is me changing the HA data retention from the default 10 to 31 days about 4 or 5 days ago but I can't remember if that coincided with the problem I'm describing. The chart is now showing 14 days of Tariff Comparison data. Maybe the later chart HTML/CSS or whatever is only expecting 10 days. Maybe that's it. However it does seem odd that nobody else has raised the issue.

gcoan, 2025-09-25:
Just reducing the retention to 14 days won't immediately do anything as HA only does the database housekeeping overnight (4am)?   You'd need to manually purge the database as well.

Mine is currently set to 9 days but I can increase it to 14 to test,  but I do have quite complex and aggressive database purging per entity so might not be able to recreate things as you have.

But I'll give it a go, but will take 5 days to build the history up to see if it happens for me.

TBH days previous of 7 and 14 is not great because it only samples 2 days.  I personally prefer a bigger range of days (mine is set to 2 through to 8) to average over.

Sycamore967, 2025-09-26:
If it helps this conversation, I have a HA retention of 21 days and the comparison has never worked for me, in any PB version.  If I start the comparison it just sits there running, with a blank screen.

brickatius, 2025-09-26:
@Sycamore967 The issue here is the Tariff Comparison section not loading. In your case it sounds as if you haven't got the relevant part of your apps.yaml set up. You should have something like this at the end of your apps.yaml
```
# Tariff comparison feature
# Adjust this list to the tariffs you want to compare, include your current tariff also
# Octopus region code (https://energy-stats.uk/dno-region-codes-explained/)
octopus_region: "H"
compare_list:
  - id: "current"
    name: "Current Tariff"
#- id: 'go_agile'
#  name: 'GO import/Agile export'
#  rates_import_octopus_url: 'https://api.octopus.energy/v1/products/GO-VAR-BB-23-02-07/electricity-tariffs/E-1R-GO-VAR-BB-23-02-07-{octopus_region}/standard-unit-rates/'
#  rates_export_octopus_url: 'https://api.octopus.energy/v1/products/AGILE-OUTGOING-BB-23-02-28/electricity-tariffs/E-1R-AGILE-OUTGOING-BB-23-02-28-{octopus_region}/standard-unit-rates/'
  - id: "agile_fixed"
    name: "Agile import/Fixed export"
    rates_import_octopus_url: "https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-{octopus_region}/standard-unit-rates/"
    rates_export_octopus_url: "https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-{octopus_region}/standard-unit-rates/"
```
Uncomment any of the tariffs you want to compare.

SwiftRR, 2025-09-26:
I am running v8.25.1
If I try WebUI Compare, I get no bar chart with a message: Loading chart:
[image]

I also get warnings in WebUI log referring to GE Cloud:
[image]

As suggested, I have downgraded to Predbat 8.24.2.
WebUI Compare now works again and I didn't even relaunch it:
[image]

The GE Cloud warnings have also disappeared but I think these may have been linked to the changes introduced in v8.25.0. 

Rob

pgit2345, 2025-09-26:
Got the same issue here, "Loading chart (please wait)..." displays. 

I've not made any changes to database retention (as mentioned above). 

In the predbat yaml I compare days 6, 7 and 8 (and have done for quite some time). Haven't made any changes to the yaml config for ages either. 

Running predbat v8.25.1

Core 2025.9.4
Supervisor 2025.09.0
OS 16.2
Frontend 20250903.5

Just downgraded to predbat v8.24.2 - the chart shows now (although as per OC above, might be intermittent on the .2 release)?

brickatius, 2025-09-27:
Ok. I'll reset my retention to the default 10 days today and see what happens with v8.25.1 tomorrow. (thank you @gcoan). 

@pgit2345 would kindly confirm that you are definitely on the default 10 days and haven't got anything like this in your homeassistant/configuration.yaml
```
recorder:
  purge_keep_days: XX
```
where XX is any number greater than 10. If you haven't it blows my theory about 8.25.1 and days > 10 not rendering the chart on the 'Compare' page out of the water!

SwiftRR, 2025-09-27:
When I downgraded from v8.25.1 to v8.24.2, the compare chart worked.
In my recorder.yaml, I have the line:
`  purge_keep_days: 9`
So I am <10

In my apps.yaml, I have:
`  days_previous:
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8
`
So I am <10 on both count. I have not changed these settings for a long time.

I don't know whether this is related but I was receiving warnings for GE_cloud https://github.com/springfall2008/batpred/issues/2690#issuecomment-3340487374
In apps.yaml, I have the GE cloud data set to false but v8.25.1 appears to be looking at GE cloud data:
`
ge_cloud_data: False
  ge_cloud_serial: '{geserial}'
  ge_cloud_key: 'xxxxxxxxxxxxxx'
`
I could upgrade again to v8.25.1 and comment out the lines to see if this solves the problem.
It would be worth checking if you are also getting these GE cloud warnings with v8.25.1.

For HA, I am running:

Core 2025.9.3
Supervisor 2025.09.0
OS 16.2
Frontend 20250903.5

I am also running predai

Rob

brickatius, 2025-09-27:
So, it would appear that this issue is not related to the number of days of retention as is shown by @SwiftRR and @pgit2345. It's obvious that something changed on the 'Compare' page after v8.24.1 as to what I personally have no idea. Is this one for @springfall2008 ?

PianSom, 2025-09-27:
I am seeing the same behaviour. I have not changed retention policy for many month. I have just noticed the same error on the current version. It hasnt bothered me enough to try a downgrade.

brickatius, 2025-09-28:
UPDATE:

Having downgraded my retention period to 10 days (see image and count the number of days shown) I can confirm, along with @PianSom and @SwiftRR,  that this bug is NOT related to the retention period, however it is a bug nevertheless.

[image]

I'm certain that is something to do with the significant changes between v8.24.2 and v8.25.0

I've gone through the changelog. I'm no expert in python but I'm pretty sure that the bug is in apps/predbat/web.py. Could it be something to do with the following new lines (2172 to 2175) which were added? (I've tried to find what they replaced but can't)

```
        if self.compare_hist:
            text += self.render_chart(series_data, self.base.currency_symbols[0], "Tariff Comparison - True cost", now_str, daily_chart=False)
        else:
            text += "<br><h2>Loading chart (please wait)...</h2><br>"
```

gcoan, 2025-09-28:
thanks for all your investigation @brickatius at least we know what its not caused by now

brickatius, 2025-09-30:
I'm trying to get to the bottom of why the Tariff Comparison chart is not displaying as it should on the 'Compare' page of the WebUI. These are the lines (2230 - 2233) in the web.pv that should display the Tariff Comparison chart. 
```
if self.compare_hist:
            text += self.render_chart(series_data, self.base.currency_symbols[0], "Tariff Comparison - True cost", now_str, daily_chart=False)
        else:
            text += "<br><h2>Loading chart (please wait)...</h2><br>"
```
I had wondered if it was something to do with the 'render_chart' function being missing but all of the other instances of 'render_chart' such as on the 'Charts' page of the WebUI all work properly, so it's not that. 
This chunk of code (lines 2230 - 2233 in web.py) is identical to what was in the earlier version but I was wondering if somehow 'compare_hist' is not getting the data it requires. 

We all appreciate the work you do with Predbat Trefor (@springfall2008) and that you are a busy man, however is there any chance you could investigate this please?

Richard

scigulinsky, 2025-10-04:
Its not working for me as well also the link to the intelligent go is old and not pulling through. I tried to fix it (new link) but failed

gcoan, 2025-10-04:
> Its not working for me as well also the link to the intelligent go is old and not pulling through. I tried to fix it (new link) but failed

The igo rates were changed a while ago, I'm using:

```
    - id: 'igo_fixed'
      name: 'Intelligent GO Import/Fixed Export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/INTELLI-VAR-24-10-29/electricity-tariffs/E-1R-INTELLI-VAR-24-10-29-A/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-BB-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-BB-24-10-26-A/standard-unit-rates/'
```

which worked fine for me last night.  Change the region code (A) to match your own

discodancerstu, 2025-10-04:
Also having the same issues as @SwiftRR 

[image]

[image]

The API links appear to be ok though, copy and pasting into a browser displays the correct rates:
[image]

Log and apps attached, the log is showing warnings "Warn fetching rates during comparison of tariff.."

[apps (2).yaml](https://github.com/user-attachments/files/22703661/apps.2.yaml)
[predbat.log](https://github.com/user-attachments/files/22703660/predbat.log)

SwiftRR, 2025-10-04:
I have been experimenting with the Compare URLs.
I tried removing -BB from the Outgoing URL and Flux Import.
I then ran 'Compare now'
I now get all the table data and there are no warnings.
But I can't get GO to work!

For now I have commented out GO. I now see everything in my tariff list.
Still no chart though.

[image]

My conclusion is that Tariff URLs have changed.
No guarantee that my conclusion is correct though!

Rob

pgit2345, 2025-10-05:
You can explore the products from here

https://api.octopus.energy/v1/products/

Look for go-fix or go-var

Then copy the URL it shows into a new tab.

E.g. https://api.octopus.energy/v1/products/GO-FIX-12M-25-08-29/

Look through that page to find the right URL for usage charges in your area 

https://api.octopus.energy/v1/products/GO-FIX-12M-25-08-29/electricity-tariffs/E-1R-GO-FIX-12M-25-08-29-A/standard-unit-rates

(For area A)

gcoan, 2025-10-05:
thanks @pgit2345 for the detailed steps, I have added these to the documentation and updated the URLs to the correct ones.  Will be in my next PR to Trefor

SwiftRR, 2025-10-05:
I use this, exemplified for my current tariff and go_fixed:

```
  octopus_region: "M"

  compare_list:
    - id: 'current'
      name: 'Current Tariff'

    - id: 'go_fixed'
      name: 'GO import/Fixed export'
      rates_import_octopus_url: 'https://api.octopus.energy/v1/products/GO-FIX-12M-25-08-29/electricity-tariffs/E-1R-GO-FIX-12M-25-08-29-{octopus_region}/standard-unit-rates/'
      rates_export_octopus_url: 'https://api.octopus.energy/v1/products/OUTGOING-VAR-24-10-26/electricity-tariffs/E-1R-OUTGOING-VAR-24-10-26-{octopus_region}/standard-unit-rates/'

```
I like this as I only have to insert the region code once rather than many times.
In our Episode 7 thread, I have added all my URLs.

Rob

gcoan, 2025-10-05:
@brickatius looks like predbat 8.25.4 has fixed the compare chart not working.  Although not mentioned in the release notes, I've just upgraded and it's working fine for me

The PV and PV7 charts are also working in this release, although the legend at the top is all zero on mine ...
[image]

Updated compare URLs are in my fork of the documentation https://github.com/gcoan/batpred/blob/main/docs/compare.md

brickatius, 2025-10-05:
Hi Geoffrey, @gcoan 

I've updated to v8.25.5 and concur that the Tariff comparison - True cost chart on the Compare tab is now working as expected and is also showing the full number of days of my retention period (14).

All of the charts on the Charts tab  are now showing but like you, the legend at the top of the PV and PV7 charts is showing
 0 kW so there is still a bit of a bug there. 

I'll load all of the compare tariff URLs into my apps.yaml and will report back tomorrow. 
Richard

gcoan, 2025-10-05:
Hi Richard

Thanks for confirming and for raising this with Trefor on FB.  I have tested all the compare URL's that are in the compare doc, finding typing mistakes I had made and some errors with the eon next drive that had been there for ages.
These all work fine for me, but second pair of eyes appreciated.

Good that you are seeing the same issue with the totals not appearing, not just me then !
Maybe you can re-raise this last bit on FB?

Cheers

gcoan, 2025-10-06:
> I revisited the charts this morning and there are now values for PV Power, Forecast history and Forecast history CL but the rest are still showing 0 kW.

Likewise, same for me:

[image]

>  One thing however: it might be an idea to advise people to replace the whole of the Tariff Comparison section of the code because of the breaking change from 'octopus_region' to 'dno_region'.

Hmm, not sure when that got changed.  It wasn't me, the current (Trefor) docs version has dno_region in.  It does make sense, it is dno_region not an octopus-specific thing.
Personally I don't use this bit, I have all my URL's hard coded to region A.
But for most people, copy/paste of the whole template is easiest.

brickatius, 2025-10-06:
FB done. I didn't specifically mention Trefor - I'm sure he always sees the FB group posts.

closed 2025-10-07 by brickatius