springfall2008/batpred #2694: Predbat Webserver Crash with Error Message "500 Internal Server Error"

jrwm47, 2025-09-27:
**Describe the bug**
Since _v8.25.1 update_, Predbat webserver crashes reproducably with selections: _Chart/InDay_, _Charts/PV_, and _Charts/PV7_.
All other webservice _Chart/_ selections do not cause crash.
HA and Predbat continue running normally following crash. Only webservice is down.
Predbat Restart gets webservice up and running normally, as long as you avoid menu options listed above.

**Expected behaviour**
Everything, including webservice, had been running faultlessly for months until _v8.25.1 update_.
No recent changes to configuration parameters apart from: input_number.predbat_holiday_days_left

**Predbat version**
v8.25.1

**Environment details**
GE AIO 13k5kWh, 3k5kW Solis PV, Octopus Agile, Solcast with CL
Standard HAOS installer on dedicated rpi4-64 1.2 GB / 2 GB

**Screenshots**
see attached

**Log file**
see attached

**Predbat debug yaml file**
see attached

[predbat_debug.yaml.txt](https://github.com/user-attachments/files/22573362/predbat_debug.yaml.txt)
[predbat.log.txt](https://github.com/user-attachments/files/22573364/predbat.log.txt)
[apps.yaml.txt](https://github.com/user-attachments/files/22573363/apps.yaml.txt)
[image]
[image]

FileGo, 2025-09-27:
I think this might be a duplicate of #2677.

Lumpybd, 2025-09-29:
Can confirm - I'm getting the exact same issue since 8.25.1. I'm currently on v8.25.3 and the issue is still present. I've also commented on #2677 for visibility.