springfall2008/batpred #2695: IOG - Multiple Car Support

bfayers, 2025-09-27:
**Is your feature request related to a problem? Please describe.**
Recently Octopus started allowing you to enroll multiple cars for IOG, the latest beta versions of BottlecapDave's Octopus integration supports this, by creating the set of intelligent dispatching, target time, and charge limit for each vehicle.

Attempting to configure the current IOG settings in predbat as a list does not appear to work for `octopus_intelligent_slot` resulting in an error when trying to run the plan:

```
2025-09-27 12:56:44.228459: Error: Exception raised 'list' object has no attribute 'lower'
2025-09-27 12:56:44.230799: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 1434, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 682, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 1107, in fetch_sensor_data
    completed = self.get_state_wrapper(entity_id=entity_id, attribute="completedDispatches") or self.get_state_wrapper(entity_id=entity_id, attribute="completed_dispatches")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/predbat.py", line 274, in get_state_wrapper
    state = self.ha_interface.get_state(entity_id=entity_id, default=default, attribute=attribute, refresh=refresh)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/ha.py", line 339, in get_state
    self.db_mirror_list[entity_id.lower()] = True
                        ^^^^^^^^^^^^^^^
AttributeError: 'list' object has no attribute 'lower'
2025-09-27 12:56:44.248724: Info: record_status Error: Exception raised 'list' object has no attribute 'lower'
2025-09-27 12:56:44.248868: Error: 'list' object has no attribute 'lower'
```

**Describe the solution you'd like**
Allow configuring all current octopus intelligent values as a list to support multiple vehicles

**Describe alternatives you've considered**
Considered using template helpers in hass to create single 'combined' entities for each of these, but that seems like a poor solution when it could be handled in predbat.

Currently I'm just changing my predbat config based on which car I have plugged in.

**Additional context**
I haven't tested this, but it seems there is no reason you couldn't have more than 1 IOG vehicle charging at the same time, potentially with different slots so all should probably be taken into account when running the plan.

matttheotter, 2025-12-17:
Oh man if this is possible it would be amazing, my folks have both their EVs and PB is only seeing one of the charging cycles, I have it configured as if it was talking to Ohme but it does make planning awkward.

matttheotter, 2025-12-18:
Oh interesting.

I use charger led control which is somewhat easier but you do need to recalc the % for each vehicle.