springfall2008/batpred #2707: Help with inveter setting 5A to timed_charge_current and timed_discharge_current limiting my charge/discharge

JoachimAndersson, 2025-09-29:
I got problem that this 2 are always set to 5A limiting my charge and discharge, how can i solve that ? 

  timed_charge_current:
  - number.inverter_battery_inverter_battery_charge_current_limit
  timed_discharge_current:
  - number.inverter_battery_inverter_battery_discharge_current_limit

For this 2  it set them to 20% ,  can i adjust how low it can set them my battery support much lower?

  reserve:
  - number.inverter_battery_inverter_battery_soc_min_on_grid

  battery_min_soc:
  - number.inverter_battery_inverter_battery_soc_min_off_grid
        

also when it change state

it do the state to start the new state then end the old one so it always do 2 commands.

for example it do:
charge_start_service
discharge_stop_service

And since it do in that order it set the latest state (i solve that with my script to only to 1 command) 

app.yaml
```
pred_bat:
  module: predbat
  class: PredBat

  prefix: predbat
  timezone: Europe/Stockholm
  currency_symbols:
  - "SEK"
  - "öre"
  threads: auto
  run_every: 5
  load_filter_threshold: 60

  num_cars: 0

  num_inverters: 1

  soc_max:
  - 25.6   

  notify_devices:
  - mobile_app_sm_s921b
  
  battery_rate_max:
  - 10000   

  inverter_limit: 17000

  load_today:
  - sensor.consumed_today
  import_today:
  - sensor.energy_meter_consumed_day
  export_today:
  - sensor.energy_meter_produced_day
  pv_today:
  - sensor.total_pv_generation_today
  soc_percent:
  - sensor.inverter_battery_inverter_battery_soc
  battery_power:
  - sensor.inverter_battery_inverter_battery_power
  pv_power:
  - sensor.total_current_pv_power
  load_power:
  - sensor.current_consumed
  grid_power:
  - sensor.net_grid_power_flow
  
  battery_voltage:
  - sensor.inverter_battery_inverter_battery_voltage
 
  
  inverter_limit_charge:
  - 10000
  inverter_limit_discharge:
  - 10000
  
  timed_charge_current:
  - number.inverter_battery_inverter_battery_charge_current_limit
  timed_discharge_current:
  - number.inverter_battery_inverter_battery_discharge_current_limit

  pv_forecast_today: sensor.pv_forecast_today_sum
  pv_forecast_tomorrow: sensor.pv_forecast_tomorrow_sum
  pv_forecast_d3: sensor.pv_forecast_day_2_sum
  pv_forecast_d4: sensor.pv_forecast_day_3_sum

  # -- Energy Rates (Energi Data Service) --
  metric_energidataservice_import: "sensor.energi_data_service_buy_price_15min"
  metric_energidataservice_export: "sensor.energi_data_service_15min"

  # Set your daily standing charge in SEK if you have one.
  metric_standing_charge: 0

  # -- Prediction & Planning Settings --
  days_previous:
  - 1
  - 2
  - 7
  - 14
  - 21
  
  days_previous_weight:
  - 0.7
  - 0.7
  - 1
  - 0.5
  - 0.5

  forecast_hours: 48

  battery_carge_power_curve:
    91: 0.91
    92: 0.82
    93: 0.72
    94: 0.62
    95: 0.52
    96: 0.43
    97: 0.33
    98: 0.24
    99: 0.24
    100: 0.24

  reserve:
  - number.inverter_battery_inverter_battery_soc_min_on_grid

  battery_min_soc:
  - number.inverter_battery_inverter_battery_soc_min_off_grid
        
  
  # Try custom inverter
  inverter:
    name: "Solinteg"
    has_rest_api: False
    has_mqtt_api: False
    has_service_api: True
    output_charge_control: "current"
    has_charge_enable_time: False
    has_discharge_enable_time: False
    charge_control_immediate: False
    current_dp: 1
    has_target_soc: False
    has_reserve_soc: True
    has_timed_pause: False
    charge_time_format: "S"
    charge_time_entity_is_option: False
    soc_units: "%"
    num_load_entities: 1
    has_ge_inverter_mode: False
    time_button_press: False
    clock_time_format: "%Y-%m-%d %H:%M:%S"
    write_and_poll_sleep: 2
    has_time_window: False
    support_charge_freeze: False
    support_discharge_freeze": False
    has_idle_time: False
    can_span_midnight: True

  charge_start_service:
    - service: script.set_inverter_mode
      mode: UPS
      log_reason: 'charge_start_service'
      repeat: True
  charge_stop_service:
    - service: script.set_inverter_mode
      mode: General
      log_reason: 'charge_stop_service'
      repeat: True
  discharge_start_service:
    - service: script.set_inverter_mode
      mode: Economic
      log_reason: 'discharge_start_service'
      repeat: True
  discharge_stop_service:
    - service: script.set_inverter_mode
      mode: General
      log_reason: 'discharge_stop_service'
      repeat: True
  discharge_freeze_service:
    - service: script.set_inverter_mode
      mode: General
      log_reason: 'discharge_freeze_service'
      repeat: True
```
    

Here is the script for my inverter if anyone wants it.

```
fields:
  mode:
    name: Working Mode
    description: The desired working mode for the inverter.
    required: true
    selector:
      select:
        options:
          - General
          - Economic
          - UPS
          - PeakShift
          - Feed-In
          - Off-Grid
          - ToU
  log_reason:
    name: Log Reason
    description: (Optional) A reason for the action, for logging purposes.
    selector:
      text: null
    default: default
sequence:
  - variables:
      stop_reasons:
        - charge_stop_service
        - discharge_stop_service
  - condition: template
    value_template: >-
      {{ (now() - (state_attr('script.set_inverter_mode', 'last_triggered') |
      default(as_datetime(0)))) > timedelta(seconds=20) or log_reason not in
      stop_reasons }}
  - choose:
      - conditions:
          - condition: template
            value_template: >-
              {{ mode not in ['General', 'Economic', 'UPS', 'PeakShift',
              'Feed-In', 'Off-Grid', 'ToU'] }}
        sequence:
          - data:
              title: Invalid Inverter Mode Provided
              message: >-
                '{{ mode }}' is not a valid working mode. Please use one of the
                allowed options.
            action: persistent_notification.create
    default:
      - variables:
          previous_mode: "{{ states('select.inverter_battery_inverter_working_mode') }}"
      - condition: template
        value_template: "{{ previous_mode != mode }}"
      - target:
          entity_id: select.inverter_battery_inverter_working_mode
        data:
          option: "{{ mode }}"
        action: select.select_option
      - delay:
          seconds: 5
      - choose:
          - conditions:
              - condition: template
                value_template: >-
                  {{ states('select.inverter_battery_inverter_working_mode') ==
                  mode }}
            sequence:
              - data:
                  title: ✅ Inverter Mode Changed Successfully
                  message: >-
                    Reason {{ log_reason }}. Inverter working mode was changed
                    from **{{ previous_mode }}** to **{{ mode }}**.
                action: persistent_notification.create
        default:
          - data:
              title: ❌ Failed to Change Inverter Mode
              message: >-
                Reason {{ log_reason }}.  Attempted to change inverter mode to
                **{{ mode }}**, but the change failed. The current mode is still
                **{{ states('select.inverter_battery_inverter_working_mode')
                }}**.
            action: persistent_notification.create
mode: single
```

gcoan, 2025-09-29:
I don't know if it's the way you pasted your apps.yaml into this issue or not but your yaml is not correctly indented 

Eg

You have 

```
  inverter_limit_charge:
  - 10000
  inverter_limit_discharge:
  - 10000
```

When it should be 

```
  inverter_limit_charge:
    - 10000
  inverter_limit_discharge:
    - 10000
```

The indentation is critical 

Also battery_charge_power_curve is spelt wrongly in the apps.yaml

gcoan, 2025-09-29:
I think I have found the error.  You have not configured your apps.yaml correctly for your inverter.

In the logfile I can see:

```
2025-09-29 21:00:36.510176: Inverter 0 SOC: 6.14kW 24% Current charge rate 2600W Current discharge rate 2600W Current battery power 1413.0W Current battery voltage 516.1V Grid power 190.0W load power 1225
```

Charge rates of 2600W are the default rates that predbat assumes for GivEnergy inverters if the inverter type is not known.
 	
This is confirmed in the logfile:

```
2025-09-29 21:00:36.509746: Inverter 0: Type GE Solinteg
```

Inverter type is GE.

In your apps.yaml you have defined:

```
  # Try custom inverter
  inverter:
    name: "Solinteg"
    has_rest_api: False
    has_mqtt_api: False
    has_service_api: True
```

But if you read the setup instructions https://springfall2008.github.io/batpred/inverter-setup/#i-want-to-add-an-unsupported-inverter-to-predbat

"Set inverter_type in apps.yaml to match the custom inverter definition ('MINE' in the example below)

  inverter_type: MINE
  inverter:
    name: "My Shiny new Inverter"
    has_rest_api: False
    has_mqtt_api: False
    output_charge_control: "power"
...
```

you have not defined the inverter_type so it is defaulting to GivEnergy and thus ignoring the max charge and discharge rates.

Plus the battery_voltage problem mentioned above.

I will update the documentation to be clearer about how to set battery_voltage as it's not currently documented.

Once you get this working can you share the complete config and I can add it to the documentation for others

JoachimAndersson, 2025-09-30:
Thanks alot it works now (I actually thought I ha set the inverter_type but obesely not :)) Now it sets the A correctly, will check all my other things as well today