springfall2008/batpred #2708: Home assistant green not accessible after the latest solcast update

mitulgovind, 2025-09-30:
Hi,

After i updated the latest solcast update, my home assistant became inaccessible! no local access or web, Nabu case login shows its connected.

please advice

gcoan, 2025-09-30:
> Hi,
> 
> After i updated the latest solcast update, my home assistant became inaccessible! no local access or web, Nabu case login shows its connected.
> 
> please advice 

Solcast update or predbat update?

You've logged this on predbat, but either way without more diagnostics it's impossible to help you.

Plenty of people have upgraded to the latest solcast and predbat versions and not experienced any problems.

I would try powering off and on again your HA Green to see if that recovers it, otherwise you'll have to recover from a backup

mitulgovind, 2025-09-30:
> Hi
>
> Thanks for the reply, I can not log in to the web or mobile app, plus my Nabu Casa account shows it not connected to cloud!
>
> My router shows the write IP address for HA.
>
> So I can’t get the logs of it.
>
> Unless there is a way to see it.
>
> Kind regards,
>
> Mitul Govind.

It is  Solcast update

mitulgovind, 2025-09-30:
everything was working fine until i did the solcast update this morning.

mitulgovind, 2025-09-30:
Maybe redo the install and upload from Nabu Casa the saved backup

mitulgovind, 2025-09-30:
> It may not be the Solcast update, just the restart of HA. Had you made other changes to any config files and you'd not restarted since doing that?

no nothing new.

closed 2025-12-09 by mitulgovind